dresden-elektronik/deconz-rest-plugin #1975: Device Support Request HEIMAN Smart metering plug HS2SK-E-FR

laurent941, 2019-10-16:
Hi

Could you add support for  HEIMAN Smart metering plug HS2SK-E-FR ( European /french version )
it is currently recognized as an  light image in phoscon but the information goes up well in home assistant

[image]

[image]

[image]

[image]

[image]

[image]

thank you in advance

laurent941, 2019-11-18:
no answer ?

closed 2020-03-24 by stale[bot]

twobeass, 2020-12-09:
Still not sending Power Infos via REST API... Possible to implement ?

SwoopX, 2020-12-09:
What is your request to look for it?