dresden-elektronik/deconz-rest-plugin #1976: Soft Start Dimming?

bensuffolk, 2019-10-17:
I notice that whenever I turn a light on (or off) it has a slight fade up to the required brightness (or fade off). I wondered if that was just me being unlucky with the bulbs I have picked (i.e is it built into the lights firmware), or if its something that deconz is doing instead of instantly snapping to the required brightness?

The fade is less than half a second or so, but it gives the feel of the light not responding instantly and so if its is  deconz feature, it would be great to have a way to turn it off.

For what its worth I have a couple of different style Ikea bulbs and sone Chinese RGB down-lighters.

bensuffolk, 2019-10-17:
Perfect, Where can I change that option?

I have looked in the Phoscon App and can't see it anywhere, also looked in the main deconZ GUI.

Probably a simple thing I'm missing!

Smanar, 2019-10-17:
Nope, from my memory there is a secret trick ^^, if you use the api, deconz don't use cluster 0006 for on/off but Level control 0008 with "move to level" command, and these one have the transition time edit box
But if you use transition time = 0, deconz will use the cluster 0006

bensuffolk, 2019-10-17:
Hmm,

Ok I'm using a a component on Home assistant that does all the talking to the API. So I will have to see what can be done to implement it in that case.

I was hoping there would be a default setting I could change.

closed 2020-02-21 by stale[bot]