dresden-elektronik/deconz-rest-plugin #1977: Request Device Support - Xiaomi A6121 Vima Smart Lock

davidramosweb, 2019-10-17:
Hello,
Could you add support for Xiaomi A6121 Vima Smart Lock lumi.lock.v1
Thanks.
https://www.zigbee2mqtt.io/devices/A6121.html
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

TTP1106, 2019-11-25:
I also recommend to add this device. it's a good alternative of danalock v3. :)

wuppi83, 2020-01-21:
Please add the device :0)

wuppi83, 2020-01-28:
Some news :D

closed 2020-06-05 by stale[bot]

pollinolas, 2020-08-20:
Hi, nothing new about Vima smart lock no deconz?
Regards

Smanar, 2020-08-20:
According to z2m, you can't use this device to open/close door.
It s just a sensor, that give some information.

pollinolas, 2020-08-20:
> According to z2m, you can't use this device to open/close door.
> It s just a sensor, that give some information.

Thanks for answering, if I know that, I have already read it in the forums, but my question is if it can be added even as a sensor to home assistant through deconz greetings again and thanks

Smanar, 2020-09-12:
Hello.
On my side I m justy trying to report the same value the device send to the Xiaomi application.
The encoding is only usefull to add new key I presume ? It s done by the Xiaomi application too, and using zigbee ? Because I don't see those feature on the capture.

Nemesis24, 2020-09-24:
Please add the device :0)

Smanar, 2020-09-24:
I have started the code https://github.com/Smanar/deconz-rest-plugin/tree/divers
But need someone with the device and an Unix machine.

Mimiix, 2020-09-25:
@Smanar Do we need to re-open?

Smanar, 2020-09-25:
IDK, for the moment I m ready but waiting for a "candidate" ^^, and can take long time

Nemesis24, 2020-09-25:
If i can help I'm with a rasbee1  on rasberry pi 3b+ i'm using zha on hassio with zigpy ... zigpy deconz...zha device handlers

Smanar, 2020-09-25:
:( Not possible recompile code on Hassio, and zigpy don't use deconz only the conbee.

pollinolas, 2020-11-13:
> I have started the code https://github.com/Smanar/deconz-rest-plugin/tree/divers
> But need someone with the device and an Unix machine.

Hi, no news about that?

Smanar, 2020-11-13:
No-one to test it, so the code is still in waiting mode.

pollinolas, 2020-11-13:
what do you need to test it?

Smanar, 2020-11-13:
To have a working code, I haven't the device to make test on my side.

pollinolas, 2020-11-13:
I have the device and a unix machine now, how I can help?

pollinolas, 2020-12-04:
> To have a working code, I haven't the device to make test on my side.

could you give some instruction to be able to try to add it to phoscon?

Smanar, 2020-12-04:
Oups sorry, haven't see you answer.
I m searching the code, and not find it yet. I will make a new one for tonight (or found the previous one)

Smanar, 2020-12-04:
OK, so sorry, I have realy loose the code.

You can take a look here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So it will be for you

```
sudo apt install deconz-dev
git clone --branch vimar https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

For the moment the code will just create a door sensor.

After that you can return to deconz, enable log, and use the door lock, I need logs starting with "extract Xiaomi special attribute", and you need info_L2 to have them.