dresden-elektronik/deconz-rest-plugin #1979: Bind a light to (existing) group in deCONZ GUI (via VNC)

JBS5, 2019-10-18:
I'm trying to bind a light to a (exsisting) group created by a Hue Dimmer Switch using deCONZ GUI (via VNC). Based on https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Pairing-Devices this should be possible using the 'Bind Dropbox' panel, but I think I do not understand how to use that. No matter what I select (the Hue Dimmer Switch or a light), I am not able to do anything in that panel. Could anyone point to in the right direction how to use this part of the deCONZ GUI?

[image]

(I am asking this because one of the 8 GU10 lights which are in a group went offline and I had to pair this one again. Adding this newly paired light to the existing Hue Dimmer group in the Phoscon add did not result in switching on/off together with the other 7. So I guess the binding is not set for this new added light).

closed 2020-02-10 by JBS5

ottelo9, 2022-08-15:
@JBS5 
I can't get it to work :(
I want to directly bind my Hue Dimming Switch to a bulb. I open the Bind Dropbox.

Source = "Hue Dimming Switch" -> "0006 OnOff" cluster (grey)
Destination = "tradfri bulb GU10" -> "0006 OnOff" cluster (blue)

click on bind = success
But I cannot switch the light on with my dimming switch (press I on the remote). Nothing happens.

[image]
[image]
[image]

JBS5, 2022-08-15:
@ottelo9 I am not using deCONZ anymore so hopefully anyone else can help you with.

ottelo9, 2022-08-16:
@JBS5 oh ok. Why?

JBS5, 2022-08-16:
@ottelo9 I was curious about Zigbee2MQTT.

ottelo9, 2022-08-16:
@JBS5 ok thx. Are you happy with z2m? Whats about ConbeeII updates and the killing feature with the vnc deconz view.

JBS5, 2022-08-16:
@ottelo9 Yes. I am not using the Conbee anymore but switched to a Sonoff Zigbee Dongle https://sonoff.tech/product/diy-smart-switch/sonoff-zigbee-dongle-plus-efr32mg21/

I guess this is not the best place to discuss the benefits of Z2M ;-)

ottelo9, 2022-08-16:
:D I know, this was the last question ;)