dresden-elektronik/deconz-rest-plugin #1980: huge load by constantly restarting

daniel-lo, 2019-10-18:
please look here .. I have no Idea what's going on.

https://photos.app.goo.gl/NLCpuNymZzqqxnxR7

Help!

P.S. I have some glitches but generally all functionality is given.

daniel-lo, 2019-10-18:
sudo systemctl stop deconz-gui -> CPU usage down to 5% in idle
gui is still accessible/connected - and so is phoscon
systemctl list-unit-files | grep enabled -> lists "deconz-gui.service" as enabled

I am really fishing in the dark since I am not much into Linux .. but whatever I did seemed to help a lot. I am still concerned though because I assume what I did is not sticky and the situation will return again. Any knowing/helpful hand is much appreciated.

daniel-lo, 2019-10-18:
Ultimately I removed the service from systemctl:
sudo systemctl disable deconz-gui
no more deconz-gui.service in the list .. and still all is good after a reboot.

ChrisHae, 2019-10-24:
So did you solve your problem?
Disabling the deconz-gui service should prevent it from start after reboot. But of course it is not usable anymore. Except you have enabled deconz.service (without gui) or you start deCONZ via command line call "deCONZ" or via the start menu link.

daniel-lo, 2019-10-24:
@ChrisHae .. if I recall correctly I have rebootet and still had the GUI operational. But I will have to validate that .. and get back to you.

daniel-lo, 2019-10-24:
@ChrisHae 
You were perfectly right .. I did have to execute:
sudo systemctl enable deconz-gui
to get it up and running after a reboot.
So now I'll have to monitor CPU usage again :-(

halloamt, 2019-10-30:
Try using `systemctl enable deconz` instead of `deconz-gui`. That starts a headless version of deCONZ. Also disable lxde as lxde can use many cpu cycles.

daniel-lo, 2019-10-31:
@halloamt .. CPU is ok for me now .. idling around 20%. Your suggestion wuld not work for me since I want to have the GUI and I want to access it via VNC.

Smanar, 2019-10-31:
20% is still a big value.
I m always under 5% with GUI an an homea utomation application on same machine than deconz.

daniel-lo, 2019-10-31:
@Smanar .. 20% seems to be a max value .. currently around 5 .. the bottom line is .. I can live with that .. when I started the thread it was around 98% constantly.

closed 2020-03-06 by stale[bot]