dresden-elektronik/deconz-rest-plugin #1981: Tradfri Light: Remember last state

m50434, 2019-10-18:
Hi everybody. I have a bulb from ikea tradfri. If I turn the light off hard and turn it on again, the light unfortunatelly forgots the last state. It always starts in a default state after the light was turned off hard.

Is there a way to remember the last state?!

Flurkmark, 2019-10-27:
Mine remembers state by default. They are one or two years old. Strange..

Smanar, 2019-10-27:
@Flurkmark it's strange, ikea bulb have this option on firmware, but not working, this option is reset at every power off, so it can't work.
But it works on philips (with recent firmware)

m50434, 2019-10-27:
In meantime I bougth some of these spots: Ikea Tradfri GU10.
Suprisly they remember the state.
The bulbs I mentioned above are the color bulbs... and they still forgot the state.

Smanar, 2019-10-27:
So you can try to change the option in deconz "poweron onoff" but from my memory this option don't work on ikea, at least on model I have tested.

m50434, 2019-10-27:
where is this option?! I didnt find any option at all...

Smanar, 2019-10-27:
On deconz, for exemple, select the cluster on/off and look at the attributes on bottom.

m50434, 2019-10-27:
Hmm ok. Im using the phoscon app in combination with the conbee usb-stick. Obviosly there are no such preferences...

Smanar, 2019-10-27:
This option is in the deconz app (with GUI).
[image]

jdeluyck, 2020-01-16:
I'm also seeing this with Tradfri bulb E14 WS opal 400lm lamps. All of them, weirdly enough.

Before linking them to Deconz i had them via the standard tradfri remote, and then they do keep their brightness setting. Now they revert.

I've tried changing that parameter but i get a MALFORMED_COMMAND back.

m50434, 2020-02-26:
Me again.
I found a way to access to the deconz GUI.
When I try to change the attribute from "On" to "Previous" I got an error => "malformed_command".
So unfortunatelly I have still the same problem with the IKEA bulbs: If they turned off hard, they cannot remember their last state :-(

zladukas, 2020-02-26:
You can use deconz-cli, just send:

zclattr <short addr> 1 0x0006 02034030FF 

02 03 4030 FF
 |  |  |   |
 |  |  |   previous  0xFF
 |  |  8 bit enumeration 0x30
 |  4003 PowerOn OnOFF attr
 write attr

eg: zclattr 0x12AB 1 0x0006 02034030FF

m50434, 2020-02-27:
Thanks for the answer.
At the moment I have no idea how to use deconz-cli through docker...

Furthermore I suggest that the error-code ("malformed_command") means, that the value "previous" is not supported by the IKEA bulbs.

Or is somebody out there with IKEA bulbs, which remember last state after turn off hard?!

zladukas, 2020-02-27:
Few seconds before my comment I’ve changed this setting for Ikea and Philips bulbs using cli plugin...

m50434, 2020-02-27:
Ok, an does it work (Remember last state) for E27 (!) bulbs ?!

zladukas, 2020-02-27:
I have no e27 Ikea, for those new fw is released. Philips yes.

m50434, 2020-02-28:
Hey thanks.
I updated the IKEA E27 color blubs to fw 1.3.0.0.9 and now (I cant believe) it works :-) => the bulbs remember there last state (after hard turn off).
GREAT !!!

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix