dresden-elektronik/deconz-rest-plugin #1982: Different URLs lead to different Gateway versions

daniel-lo, 2019-10-18:
Via my LAN IP http://192.168.178.114:8080/ Phoscon shows version 2.05.67 / **26.8.2019**
Via the generic  http://dresden-elektronik.de/pwa/ my Gateway shows Version 2.05.67 / **4.10.2019**

Same Gateway. How come?

salopette, 2019-10-18:
me to

ebaauw, 2019-10-18:
Same gateway version, 2.05.67; different version of the Phoscon web app.

daniel-lo, 2019-10-18:
Product - Version - Firmware .. how should one know that this is related to the Phoscon app version?
Also I read that everything stays local even if you use that external link - so I expected it to call my local phoscon app  .. and no external one. I am confused .. clarification would be much appreciated.

Ultimately: Is it better to use the DE-Link because the version is newer?

ebaauw, 2019-10-18:
The gateway runs locally on your server.  The Phoscon app runs locally in your web browser.  It uses the REST API to talk to the gateway.  You can load the Phoscon app from your local server (the version included in the deCONZ package), or from the dresden elektronik web site (the latest version).

daniel-lo, 2019-10-19:
thank you .. so my interpretation is: use the one from d-e web site then you may benefit from the newest features or fixes.

closed 2020-02-23 by stale[bot]