dresden-elektronik/deconz-rest-plugin #1986: ConBee2 does not work on a Raspi4 if USB SSD is connected to USB3 port

sebastianlorenzen, 2019-10-19:
Hello, I moved my phoscon conbee2 installation from an old raspi to a new raspi4 with debian buster.
It did not run. In the phoscon app the stick allways showed as "Firmware not connected"

I got it to work after I plugged my USB SSD into an USB2.0 slot. When I move it back to USB3.0 the conbee2 stops working again.

Is this known? Can someon reproduce this?

rmkew377, 2019-10-22:
Hi,
it is well known that USB3 devices can affect WiFi (ZigBee is in the same frequnency range) and even DECT:  
https://avm.de/service/fritzbox/fritzbox-7490/wissensdatenbank/publication/show/1636_WLAN-oder-DECT-Storungen-bei-angeschlossenem-USB-3-0-Gerat/

Is the ConBee Stick directly plugged into the RasPi USB-Port? 
Maybe it helps, if the Conbee - Stick (and the SSD) is connected via a USB extension, to inlarge the  distance between SSD(-Cable) and ConBee. Using high quality shielded cables for both devices is mandatory.

sebastianlorenzen, 2019-10-22:
That could be, but doesn't matter as the conbee stick is not recognized by phoscon correctly. So must be something internal.

sebastianlorenzen, 2019-11-04:
I have tested a bit arround and a 1meter long cable helped at the end. Strange, that the stick is recogniced by the OS, but not from Phoscon App. With the long cable all works fine. (At least since 3 hours)

closed 2019-11-04 by sebastianlorenzen