dresden-elektronik/deconz-rest-plugin #1987: RaspBee stuck in firmware update loop / cannot update manually

stuhlpfarrer, 2019-10-19:
Hi,
I have a similar problem as https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1712
When I start the dconz application it connects to the device, reports the need to update the firmware, progresses till 100% then reconnects, ...
I have updated the dconz app manually as described here:

> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian

> [Optional] Update the deCONZ package as well.
After fetching the most recent binary from here https://www.dresden-elektronik.de/rpi/deconz-firmware/deCONZ_Rpi_0x26330500.bin.GCF

this happens:
```
$ sudo GCFFlasher_internal -f deCONZ_Rpi_0x26330500.bin.GCF
RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
RE (586 ms)
 premium
Vers. 1.02
build 2013/08/01

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
REA (586 ms)
 premium
Vers. 1.02
build 2013/08/01

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
READ (586 ms)
flashing 125684 bytes: |==error: timeout flashing firmware after 3001 ms
```

Please support.

bdickhaus, 2019-10-22:
Have you tried power cycling the RaspBee system? That fixed a similar problem for me.

stuhlpfarrer, 2019-10-22:
Thanks for the tip but yeah I did that 3-4 times already (with the linux package installed) then couple more times with the newest complete sd card image. Always the same issue.

Smanar, 2019-10-30:
Hello, are you sure deconz is closed ?

`ps ax | grep deconz`

stuhlpfarrer, 2019-11-09:
I have checked whether deconz is running (ps aux and pgrep), stopped both the application as well as the gui but the result is the same. 

flashing 125684 bytes: |==error: timeout flashing firmware after 3001 ms

gutoruts, 2020-02-19:
Just faced this problem. To solve it (at least update the firmware and stop getting the notification), I placed the Raspbee on my Raspberry 3B+ with an ethernet cable with internet connection, a hdmi monitor and a SD card with the following image found on the Phoscon website:
"Raspbian Buster Desktop - Stabiles Image mit Desktop basierend auf Debian Buster."
You just have to power on and watch it. The system will boot, the software starts automatically and the firmware update progress bar starts running. No prompt, no user input, no nothing. Just watch it.
When back to my ohter configuration the firmware is listed as up to date.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix