dresden-elektronik/deconz-rest-plugin #1989: How to connect VNC on Raspberry Pi?

ir-fuel, 2019-10-20:
I have a Raspberry Pi (without screen nor keyboard) and would like to connect to the app using VNC.

I installed VNC Viewer but the connection is refused.

I tried:

```
$ sudo systemctl disable deconz
$ sudo systemctl stop deconz

$ sudo systemctl enable deconz-gui
$ sudo systemctl start deconz-gui
```

But that gives errors.

```
Oct 20 15:29:06 phoscon systemd[1]: deconz-gui.service: Main process exited, code=exited, status=1/FAILURE
Oct 20 15:29:06 phoscon systemd[1]: deconz-gui.service: Failed with result 'exit-code'.
Oct 20 15:29:06 phoscon systemd[1]: deconz-gui.service: Service RestartSec=100ms expired, scheduling restart.
Oct 20 15:29:06 phoscon systemd[1]: deconz-gui.service: Scheduled restart job, restart counter is at 3.
Oct 20 15:29:06 phoscon systemd[1]: Stopped deCONZ: ZigBee gateway -- GUI/REST API.
Oct 20 15:29:06 phoscon systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Oct 20 15:29:06 phoscon deCONZ[495]: qt.qpa.screen: QXcbConnection: Could not connect to display :0
Oct 20 15:29:06 phoscon deCONZ[495]: Could not connect to any X display.
Oct 20 15:29:06 phoscon systemd[1]: deconz-gui.service: Main process exited, code=exited, status=1/FAILURE
Oct 20 15:29:06 phoscon systemd[1]: deconz-gui.service: Failed with result 'exit-code'.
Oct 20 15:29:06 phoscon systemd[1]: deconz-gui.service: Service RestartSec=100ms expired, scheduling restart.
Oct 20 15:29:06 phoscon systemd[1]: deconz-gui.service: Scheduled restart job, restart counter is at 4.
Oct 20 15:29:06 phoscon systemd[1]: Stopped deCONZ: ZigBee gateway -- GUI/REST API.
Oct 20 15:29:07 phoscon systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Oct 20 15:29:07 phoscon deCONZ[498]: qt.qpa.screen: QXcbConnection: Could not connect to display :0
Oct 20 15:29:07 phoscon deCONZ[498]: Could not connect to any X display.
Oct 20 15:29:07 phoscon systemd[1]: deconz-gui.service: Main process exited, code=exited, status=1/FAILURE
Oct 20 15:29:07 phoscon systemd[1]: deconz-gui.service: Failed with result 'exit-code'.
Oct 20 15:29:07 phoscon systemd[1]: deconz-gui.service: Service RestartSec=100ms expired, scheduling restart.
Oct 20 15:29:07 phoscon systemd[1]: deconz-gui.service: Scheduled restart job, restart counter is at 5.
Oct 20 15:29:07 phoscon systemd[1]: Stopped deCONZ: ZigBee gateway -- GUI/REST API.
Oct 20 15:29:07 phoscon systemd[1]: deconz-gui.service: Start request repeated too quickly.
Oct 20 15:29:07 phoscon systemd[1]: deconz-gui.service: Failed with result 'exit-code'.
Oct 20 15:29:07 phoscon systemd[1]: Failed to start deCONZ: ZigBee gateway -- GUI/REST API.
```

Any idea how to solve this?

ir-fuel, 2019-10-20:
Fixed.
Had to follow this:

https://www.raspberrypi.org/documentation/remote-access/vnc/

and

https://www.raspberrypi.org/forums/viewtopic.php?t=216737

closed 2019-10-20 by ir-fuel

hesspoint, 2021-05-05:
Hi,
I have the issue that I have a Pi with a Raspbee II and the Pi is headless.

I can connect to the Pi via SSH and also via VNC. I can start Deconz on the command line and I can access the Phoscon app, add devices etc. everything fine.

However I want to open the Deconz app to be able to see the signal strength of the devices.

So I connect to VNC  and then open Programming > Deconz. And nothing happens. I thought the app is broken but when I connect the Pi to a screen and keyboard then I can open the Deconz app without any problem. But not in VNC. That is super strange.

Also I cannot use that way because I need to move the Pi to a different room and I need to to see the signal strength from the location where it is normally located. So VNC is the only option.

AdamBremholm, 2021-05-13:
> Hi,
> I have the issue that I have a Pi with a Raspbee II and the Pi is headless.
>
> I can connect to the Pi via SSH and also via VNC. I can start Deconz on the command line and I can access the Phoscon [...]

exact same here! did you manage to find any solution for this problem? I can start the gui by typing 
sudo systemctl start deconz-gui in command line but I cant get it to connect. When i press deconz app in start menu nothing shows up.

hesspoint, 2021-05-14:
Yes I got some great tips here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4839