dresden-elektronik/deconz-rest-plugin #1990: All kinds of issues with connected devices

ir-fuel, 2019-10-20:
In summary. All the below used to work but now it's acting up and I have no idea on how to debug this.

- I have 2 Aqara switches. One shows as connected, one is grayed out. The one that is connected does not react to button presses

- I have 2 Aqara sensors. Both show as connected but none of them react when I open/close the window they are attached to

- I have 1 Hue light. It shows connected but not matter what I do in the GUI, the light never changes color/intensity/...

It's driving me insane. The first issue I had was with the light. Then I power cycled the Raspberry Pi and now everything seems to act up.

Any idea on how to solve this?

[image]
[image]

ir-fuel, 2019-10-20:
I also do not have any connections
[image]

Smanar, 2019-10-20:
Hello,
All your devices are disconnected, grayed name.
You have already see somes connections on the GUI or never ?
Have you tried to re-pair your device ? At least the HUE bulb.

ir-fuel, 2019-10-20:
I first deleted the Hue bulb. Now I deleted a switch, factory reset it and added it again. Now it shows up with a blue line to the coordinator.

But why did everything go offline? Is it because a Hue bulb has a different role in the network?

ir-fuel, 2019-10-20:
Also, my windows sensors are shown grayed in the VNC view, but in the web view they show as connected.

ir-fuel, 2019-10-20:
I am clueless now. I reinstalled my Raspberry pi (wrote a new image on the SD card to start from scratch, using Balena Etcher) and I can't even add a window sensor that was never paired before.

Smanar, 2019-10-20:
The hue bulb can act as router, but it s a device easier to include, Xiaomi device can have more problem.
You are using raspbee ?

For Xiaomi sensor, if you have problem, there 2 tricks that works, during the pairing procedure, after the reset ( = press 8/10s up the led blink 3 times) :
- Make it working, like use the magnet for windows sensor
- Press the button shortly every 1/2s.

ir-fuel, 2019-10-20:
Yes I am running on a Raspberry Pi 3+

ir-fuel, 2019-10-20:
And now while trying to pair a switch all of the sudden this appeared, but the website still says it's searching for new switches and nothing happens.
[image]

There also is a dot on the left that flashes green and sometimes red.

ir-fuel, 2019-10-20:
I just ran 

`deCONZ --http-port=80 --dbg-info=1 --dbg-error=1 --dbg-aps=1 --dbg-zdp=1`

Attached is the log file of when I was trying to add an Aqara door sensor.
[out.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3748249/out.txt)

Smanar, 2019-10-21:
Ah and in the mean time the 0x3724 changed to 0x32A0
It s something normal, every time a device will (try) connect, the gateway give it a new adress.

I m not pro, but as the pairing start to work (you can see it in phoson), you have red flash, and the same device try many time (a new adress a every try), I think you can try the second method.
When you try to pair your device, make the reset procedure, ana after that press every 1s shortly the button to prevent the device going in "hibernation" and don't answer to request (red flash).

closed 2020-02-27 by stale[bot]