dresden-elektronik/deconz-rest-plugin #1991: Ikea devices might show wrong state/brightness

xydix, 2019-10-20:
Hi guys.
I run deconz in docker (container from marthoc 2.05.69) and controlling everything from homeassistant.
I have some problems, somtimes my lights (IKEA colored lights) show wrong brightness.
If i change color / brightness from an automation it could get out of sync.
Example: I set brightness:1 and some color from an automation the light could be, for example, brighter than 1. The brightness could be 20% or 70% or what ever. Homeassistant show brighness: 1, Deconz (Phoscon App) show the lowest brightness but the lamp can be very bright. If I, with the slider in deconz, start to bright it up, the lamp will become darker bacause the slider is in the lowest part of the brightness slider.
IF i make just one change, first brightness, then put a delay in the automation, then change the color in at new command it seems to work better.

I also tried to create a "wakeup light" with brigness, transision time and color change. This seemed impossible, got out of sync every time.

Another way to recreate this is to use the app HomeHabit>homeassistant>deconz.
If I try to only change color from HomeHabit, sometimes the light might go to 100% brightness in the app but the light is on a low lever of brightness or vice verse.

Every time all sliders show the same, HomeHabit, HomeAssistant and Deconz show the same brightness. That is why i suspect there is something wrong with Deconz.

FelixUster, 2019-10-29:
I can confirm this issue. I have an automation that sets brightness and color temp within one call for an ikea lightbulb. But only the color temp is set on the bulb, even though both parameters are correctly display within phoscon.
If I split the setting into 2 commands (with a delay between) it works as expected. 
In my opinion it is a new issue as some weeks ago, both worked for me

noxhirsch, 2019-11-04:
Same issue here ...

barton-im, 2019-11-04:
I am having similar issues with Osram bulb. I am setting the brightness in a home assistant automation (decrease of brightness). Seems that the brightness readings are off half the time (not exactly what I set in the previous step).

helto4real, 2019-11-22:
Same issue, cant set color temp and brightness the same time.. With many lights it even fails with transition=0. 
version 2.05.71 
FW: 264A0700

Flurkmark, 2019-11-22:
@FelixUster Did you upgrade in this time, if so between which versions? 

I have always had a problem with IKEA lights and resolved to using scripts. However some small lights I have kept as scenes since they always makes small changes between scenes. I have noticed lately that they sometimes are completely wrong.

closed 2020-03-28 by stale[bot]

Gerben321, 2022-11-15:
I am facing the same problem with my Gunnarp light in the bathroom. Any update to this?