dresden-elektronik/deconz-rest-plugin #1992: Ikea on/off outlet show wrong state

xydix, 2019-10-20:
I use IKEA on/off plug far away from my gateway
I noticed, if the signal can't get throug, deconz still show the change of state that didn't happend.
I compare it with my Z-Wave network. If the same thing happend there, like i turn a switch on but the signal do not reach the outlet it will still show off.
With Deconz it will show on but the outlet is still off.
To me this feels like using old 433Mhz. Sure, the communication is more stable with mesh but if the state show on, I want it to be on, nothing else.
Is there any settings to solve this?

Smanar, 2019-10-22:
Ha yes, I have make tries with an osram plug.
If I unplug the plug, I still have websoket return with confirmation if I try on/off. The websocket return isn't based on a device response ?

xydix, 2019-11-06:
I m very suprised about this. Is all Zigbee-network like this or is it only deconz? This function must the med most basic function for a modern smart home network.

xydix, 2019-11-06:
I  checked the "deconz zigbee mesh app". There it show the right state all the time. Shouldn't phoscon/REST API get info from there?
Like: Turn on outlet> check response > response? / no response? > state? > refresh state to actual state.

closed 2020-03-13 by stale[bot]

oyta, 2020-11-21:
I also have this problem.  I have several Ikea plugs, and I experience the same as @xydix . Some times Phoscon reports on, while it is off, and the other way. I tried to log in to VNC on Phoscon, but it seems its not installed on the image  I downloaded from dresden. Are there any relevant logs I can send you from Phoscon-UI?

This problem is serious, because I manage my heater with them. In the middle of the night, the room temp is suddenly 25 degrees C, while it should be 18. Or it is suddenly 16 while it should be 18.

Smanar, 2020-11-21:
Long story, but too long to explain in a closed issue.
To make short, this is "normal", the bad state is corected some time after, I think you have problem with reporting or connexion with the plug.

Same issue than https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3605