dresden-elektronik/deconz-rest-plugin #1994: Can not disable wifi in Phoscon

daddvok, 2019-10-21:
Conbee Wi-Fi is not configured, but ssid Phoscon-Gateway-B230 is visible 
I can’t connect to my network, conbee does not see it
But I also can’t disconnect Wi-Fi from the Phoscon, otherwise all network settings fall, including eth0

version 2.05.69/06.09.2019
firmware 26330500
[image]
[image]

ChrisHae, 2019-10-24:
If you want to disable phoscon to configure your wifi disable the deconz-wifi.service with:
sudo systemctl stop deconz-wifi and sudo systemctl disable deconz-wifi.

Alternatively you could try to configure the Accespoint in the phoscon app. That usually solve some problems. Then disabling it in the app should work.

Or is it something else you want to do ?

daddvok, 2019-10-24:
I tried to enter commands, reboot, but nothing has changed, i still see the network active
[image]
[image]
i see wlan0, but I did not configure it

ChrisHae, 2019-10-25:
If you want do disable the wifi for good, this should definitely work: 

Add this line to the file /boot/config.txt:
dtoverlay=disable-wifi

daddvok, 2019-10-25:
I need only deconz wifi off, wifi network of conbee, but wifi on raspberry i need.

sudo systemctl stop deconz-wifi and sudo systemctl disable deconz-wifi did not solve the problem

andreyoshev, 2019-11-10:
Same.

>sudo systemctl stop deconz-wifi and sudo systemctl disable deconz-wifi did not solve the problem

ChrisHae, 2019-11-21:
systemctl stop deconz-wifi and sudo systemctl disable deconz-wifi will prevent deconz to reconfigure the wifi of the raspberry pi. Then you can configure your wifi how you like. There are several guides on the web. For example to configure a wifi accesspoint: https://thepi.io/how-to-use-your-raspberry-pi-as-a-wireless-access-point/
or connect to another wifi: https://raspberrypihq.com/how-to-connect-your-raspberry-pi-to-wifi/

ebaauw, 2020-02-22:
> systemctl stop deconz-wifi and sudo systemctl disable deconz-wifi will prevent deconz to reconfigure the wifi of the raspberry pi.

Only until the next reboot: `deCONZ-init.sh` re-enables the service on any non-Docker Raspbian installation:
  ```
  if [ $DOCKER -eq 0 ] && [ "$OS_ID" = "raspbian" ]; then
        ENABLE_SERVICE=1
  fi
  ```

Janhouse, 2020-05-22:
Had the same problem. For now I did a workaround - edit the /etc/os-release to not say it is raspbian.

Mimiix, 2020-06-12:
I'll move this over to the Phoscon repository.