dresden-elektronik/deconz-rest-plugin #1996: Lost of all network after Halt, Unplug/Plug and restart

gagadget, 2019-10-21:
Hello,

My Conbee GW is attached to my Rasberry Pi running HASSio using a USB extension cord.

Each time I reboot my PI, the Conbee GW loose the connexion to the Zigbee network : every devices appear in the VNC app but, they are not linked together and all the lights appears in grey in the conbee web app.

I tried everything I could, I am stuck.

Thx

Smanar, 2019-10-21:
Hi,
You have only xiaomi devices ?
Or you have powered devices too ?

gagadget, 2019-10-21:
Only TRÅDFRI devices : 21 lights, 2 power sockets and 3 motion sensors

bdickhaus, 2019-10-22:
Does it eventually all start working? It takes a while for deCONZ to collect network information after a restart. For my network it usually takes about 5 minutes, but it could take longer. So I do not reboot my RaspBee gateways unless I really have to.

gagadget, 2019-10-22:
Nothing, just one detector and one remote got back again after a short period of time.

gagadget, 2019-10-22:
Deleted and Readded one device. It appears on the VNC app WITHOUT any link and does not appear in the web app ...

I suspect a corrupted DB, any advise ?

gagadget, 2019-10-22:
Done a GW reset, reboot, add one device. Was working. Did a reboot. Not working.

Here is the log in HA : 

starting version 232
[11:17:45] INFO: Starting VNC server...
[11:17:50] INFO: Starting the deCONZ gateway...
[11:17:50] INFO: Running the deCONZ OTA updater...
[11:17:50] INFO: Running the IKEA OTA updater...
[11:17:50] INFO: Running Hass.io discovery task...
{"uuid":"33084ef8af0b4fb1939293838f71ca6e"}
[11:17:50] INFO: Success send discovery information to Home Assistant
[11:17:50] INFO: deCONZ is set up and running!
libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/arm-linux-gnueabihf/dri:${ORIGIN}/dri:/usr/lib/dri)
libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/arm-linux-gnueabihf/dri:${ORIGIN}/dri:/usr/lib/dri)
libpng warning: iCCP: known incorrect sRGB profile
11:17:52:730 HTTP Server listen on address 0.0.0.0, port: 40850, root: /usr/share/deCONZ/webapp/
11:17:52:836 CTRL. 3.16.211:17:53:199 dev /dev/ttyAMA0
11:17:53:199 COM: /dev/ttyACM0 / serialno: DE2126914
11:17:53:199 COM: --dev: /dev/ttyACM0 (ConBee II)
11:17:53:199 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt
11:17:53:807 parent process bash
11:17:53:807 gw run mode: docker/hassio
11:17:53:807 GW sd-card image version file does not exist: /data/.local/share/dresden-elektronik/deCONZ/gw-version
11:17:53:807 sd-card cid: 035344535033324780ffffffff01329d
11:17:53:849 DB sqlite version 3.16.2
11:17:53:853 DB PRAGMA page_count: 30
11:17:53:853 DB PRAGMA page_size: 4096
11:17:53:853 DB PRAGMA freelist_count: 0
11:17:53:853 DB file size 122880 bytes, free pages 0
11:17:53:853 DB PRAGMA user_version: 6
11:17:53:853 DB cleanup
11:17:53:856 DB create temporary views
11:17:53:973 don't close database yet, keep open for 900 seconds
11:17:53:974 started websocket server at port 40860
11:17:53:995 found node plugin: libde_rest_plugin.so - REST API Plugin
11:17:54:002 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
11:17:58:435 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
11:17:58:466 dev /dev/ttyAMA0
11:17:58:466 COM: /dev/ttyACM0 / serialno: DE2126914
11:17:58:466 COM: --dev: /dev/ttyACM0 (ConBee II)
PROTO: CRC error
PROTO: CRC error
11:17:58:548 DEV config changed event
11:17:58:931 Device firmware version 0x264A0700
11:17:58:943 unlocked max nodes: 200
11:17:59:024 Device protocol version: 0x010B
11:17:59:037 new node - ext: 0x00212effff04f365, nwk: 0x0000
11:17:59:098 don't close database yet, keep open for 900 seconds
11:17:59:099 LightNode 1: Escalier added
11:17:59:278 dev /dev/ttyAMA0
11:17:59:283 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF
11:17:59:283 GW firmware version: 0x264a0700
11:17:59:283 GW firmware version is up to date: 0x264a0700
11:17:59:287 Current channel 11
11:17:59:305 CTRL ANT_CTRL 0x03
11:17:59:336 Device protocol version: 0x010B
11:17:59:395 Current channel 11
11:17:59:414 CTRL ANT_CTRL 0x03
11:17:59:453 Device protocol version: 0x010B
11:17:59:515 Current channel 11
11:17:59:536 CTRL ANT_CTRL 0x03
11:17:59:565 Device protocol version: 0x010B
11:17:59:627 Current channel 11
11:17:59:647 CTRL ANT_CTRL 0x03
11:17:59:966 Announced to internet
11:17:59:966 discovery server date: Tue, 22 Oct 2019 09:17:59 GMT
11:17:59:966 	 local time seems to be ok
11:17:59:966 discovery found version 2.04.35 for update channel stable
11:18:04:363 don't close database yet, keep open for 900 seconds
11:18:08:303 New websocket 192.168.1.98:63856 (state: 3) 
11:18:09:375 saved node state in 0 ms
11:18:09:587 sync() in 211 ms
11:18:30:584 0x086BD7FFFE55CA3A error APSDE-DATA.confirm: 0xD0 on task
11:18:37:152 scan finished
11:18:41:489 0x086BD7FFFE55CA3A error APSDE-DATA.confirm: 0xD0 on task
11:18:52:372 Current channel 11
11:18:52:379 Device TTL 3495 s flags: 0x7
11:18:52:386 0x086BD7FFFE55CA3A error APSDE-DATA.confirm: 0xD0 on task
11:19:03:184 0x086BD7FFFE55CA3A error APSDE-DATA.confirm: 0xD0 on task
11:19:14:078 0x086BD7FFFE55CA3A error APSDE-DATA.confirm: 0xD0 on task
11:19:14:078 max transmit errors for node 0x086BD7FFFE55CA3A, last seen by neighbors 75 s
11:19:14:151 0x5495 seems to be a zombie recv errors 6
11:19:14:151 LightNode removed 0x086bd7fffe55ca3a
11:19:14:151 Node zombie state changed 0x086bd7fffe55ca3a
11:19:52:368 Current channel 11
11:19:52:377 Device TTL 3435 s flags: 0x7
11:20:52:375 Current channel 11
11:20:52:386 Device TTL 3375 s flags: 0x7
11:21:52:362 Current channel 11
11:21:52:371 Device TTL 3315 s flags: 0x7
11:21:59:373 dev /dev/ttyAMA0
11:21:59:373 GW firmware version: 0x264a0700
11:21:59:373 GW firmware version is up to date: 0x264a0700
11:22:25:855 Remove websocket 192.168.1.98:63856 after error Unknown error
11:22:26:992 New websocket 192.168.1.98:63914 (state: 3) 
11:22:52:368 Current channel 11
11:22:52:377 Device TTL 3255 s flags: 0x7
11:23:52:368 Current channel 11

Hammertime84, 2019-10-22:
I think I am experiencing the same problem.. I have only 3 Xiami Aqara/Mijia Sensors. After a reboot of Hass.io, all sensors  are grayed out in the Phoscon Web app. 

Via the VNC viewer all sensors are gone, except for the temperature/humidity sensor. But this sensor is not connected (green line) to the coordinator (Conbee)

Is this problem introduced recently for you? Because I just bought the Conbee II last week..

Smanar, 2019-10-22:
Have you try to take a look in the hidden configuration > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1849#issuecomment-530287817 to check if something happen to your network.

But it like you have connection problem, have you a USB DDR, a metal case, ... ?

gagadget, 2019-10-22:
Yes I did check the hidden conf and it was showing only one line.

Hammertime84, 2019-10-23:
> Yes I did check the hidden conf and it was showing only one line.

Same here, just one line. After updating HA to 100.3 and HassOS to 2.12 still the same problem.
I did find another problem what is possibly related. In the Hassio logging I've found the line:

ERROR (MainThread) [hassio.supervisor] Supervisor does not run in Privileged mode. Hassio runs with limited functionality!

Maybe this causes problems.. I will start to reinstall HASSIO on another RPI. I will let you know the results..

Hammertime84, 2019-10-23:
Ok, reinstalled Hassio from scratch, integrated 3 sensors, rebooted HASSIO from VNC I only see the temp/humidity sensor. Motion and door/window sensor are gone.

Smanar, 2019-10-23:
Deconz is one thing HASS is another, it's not the same thing deconz can work even HASS is broken.
How much time are you waiting after pairing and before reboot ? I noticied that xiaomi sensor have curious reactions during the firsts hours after pairing.
And xiaomi devices leave the network if the gateway or their parent are off, it's normal but it takes hours ...

schwatter, 2019-10-23:
The same for me. I have only 4 lights (tint). They work rocksolid till i reboot my pi3 with conbee stick.
Solution is always, light switch off, wait some seconds and light switch on. After that, all lights 
up and working immediately. All updates done before.

Smanar, 2019-10-23:
BTW when you have made your last tries, have you used too a direct connexion with a remote or a switch ?
cf https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261#issuecomment-545543147

Hammertime84, 2019-10-24:
> And xiaomi devices leave the network if the gateway or their parent are off, it's normal but it takes hours ...

That was it. I was too impatient. The motion and door/window sensor took their time to be visible.
Thank you!

gagadget, 2019-10-24:
Back to the original issue with the POWERED tradfri light, any advise ?

Should I try again with a debut mode on ( which I don't know if it exists ) ?

Smanar, 2019-10-24:
@gagadget Yes there is log, but I m not able to understand them. If you don't have support here, try direclty the support. But it seem you are not alone with this kind of bug.

BTW can you make a try just with bulbs, without remote and switch pls ?

dandrzejewski, 2020-01-01:
If you "reset" the devices by selecting them in deCONZ and pressing "F5" - does that restore your connectivity?  That seems to help for me.  Does not fix the actual problem, but at least it's a quick way to get things back.  It doesn't look like there's an API for that function.

TheWizz, 2020-03-24:
We've had the same problem for several months. After restarting the Deconz server, all lights fail to control. A power cycle of the lights fixes it. Note that even though Deconz can't control the lights, a Philips remote can, so there seems to be nowthing wrong with the light or the zigbee network.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix