dresden-elektronik/deconz-rest-plugin #1997: Discover not finding fully operational Deconz/Phoscon

arnonel, 2019-10-22:
Hi

I have 2 servers with Deconz/Phoscon, but http://www.dresden-elektronik.de/discover/ is only discovering one. 

Both are fully operational and working. Why is it not finding the other one? 
p.s.
The one its not finding is using Pi Image with Conbee II

arnonel, 2019-10-31:
Also cannot connect to Home Assistant because of this reason

thebluehill, 2020-02-28:
Similar problem at my raspberry since a few days:
deconz service is running (Deb 9.4)
 ```
sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; disabled; vendor preset: enabled)
   Active: active (running) since Fri 2020-02-28 12:05:23 CET; 4min 2s ago
 Main PID: 1254 (deCONZ)
   CGroup: /system.slice/deconz.service
           └─1254 /usr/bin/deCONZ -platform minimal --http-port=80

Feb 28 12:05:23 rpi20 systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Feb 28 12:05:24 rpi20 deCONZ[1254]: libpng warning: iCCP: known incorrect sRGB profile
Feb 28 12:05:25 rpi20 deCONZ[1254]: This plugin does not support propagateSizeHints()
Feb 28 12:05:25 rpi20 deCONZ[1254]: This plugin does not support propagateSizeHints()
Feb 28 12:05:55 rpi20 deCONZ[1254]: This plugin does not support propagateSizeHints()
```

Phoscon can be reached via http://ip:80 and 3rd party apps can be started, but https://phoscon.de/discover does not recognize gateways, also NodeRed with node-red-contrib-deconz v1.1.12 is not able to find the gateway.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix