dresden-elektronik/deconz-rest-plugin #2005: Explain why update/install WiringPi on RPi4?

00lex, 2019-10-26:
Hi im using deconz Raspbee since 2 Years and 3 Months on a Rpi4. 
I have never issues with my rpi4. with the last update you write we have to install the latest version von WiringPi without exception of usage reaspbee/conbee. 

Why is WiringPi needed? And is it necessary if someone use conbee stick?

thanks

schwatvogel, 2019-11-04:
I'd also like some information regarding wiringpi, especially as its unsupported now.
Gordon got to frustrated by supporting commercial products using his code.
As i am on archlinux for arm7h, i need to create my own packages from those debs (not shure if it works though), how will you proceed from here without gordons code?

manup, 2019-11-04:
The wiringpi package will be replaced by libgpiod which is available for more architectures. It will further be loaded dynamically so it is only required when RaspBee is used. Note that for ConBee II neither wiringpi nor libgpiod is needed.

No ETA yet but we already have made a successful test with RaspBee and libgpiod.

closed 2020-03-10 by stale[bot]