dresden-elektronik/deconz-rest-plugin #2006: Some question about OTA file.

Smanar, 2019-10-26:
Hello, we can use wireshark to retreive all data during an OTA firwmare update, but how to make a file with them ?
Where can we find some informations about the *.zigbee or *.ota.signed file ? about convertions, headers, ect ...

manup, 2019-11-10:
The file format is part of the standard and defined in the Zigbee specification. A detailed description can be found in http://www.zigbee.org/wp-content/uploads/2014/11/docs-09-5264-23-00zi-zigbee-ota-upgrade-cluster-specification.pdf

So basically you need to capture the data transferred in the OTA process and rebuild the OTA header based on the Query New Image request and response.

Can be done, but it's tricky :)

Smanar, 2019-11-10:
Nice, thx.
I have started, and I can read same values than deconz on an OTA file.
But I have problem to make a file, I can make an OTA header, but I don't find information to make header on the ota file.
The ota header start at offset 456 in file, and I don't find information on previous code.

manup, 2019-11-10:
In the linked document have a look in section "6.3.2 OTA Header Format". The header is not transferred via OTA, but the related values (image type, file version, etc.) can be snooped from the Query Next Image response.

Smanar, 2019-11-10:
Yep, it's the header OTA, but not the header file.
My objective if making an ota file.
For exemple in this section the first header is "OTA upgrade file identifier" and this header is at bytes 456.
And from this section the file is composed by
- OTA header
- upgrade image
- signer certificate
- signature

But what is there before the 456 bytes. The 4 first bytes of the file are "NGIS".

manup, 2019-11-10:
Seems to be some manufacturer specific wrapper with some added data, can you share the file?
For OTA to work with deCONZ you only need the data starting with "OTA upgrade file identifier" (assuming the device to update also speaks standard OTA without proprietary extensions.

Smanar, 2019-11-11:
I m making my test with the ikea one.
You have theses strange datas on all ikea firmware, but not on Osram one. But if deconz don't use them to update bulbs, I don't think theses datas are usefull, ATM I m just searching the first usefull header field and start the code at this offset.
For the moment all the firmware I have tested are realy simple (small header + only upgrade image as data, no signatures, no verifications.

Now I have 2 working python script, 1 that can read data from an OTA file and 1 to write my own one (readable in deconz, but not tested with real data)

BTW, deconz display "Min HW version" and "Max HW version" in all firmwares, but theses fileds aren't present in all firwmares.

Thx again, now the harder part, use sniffed data in the python file ^^.

closed 2020-03-18 by stale[bot]