dresden-elektronik/deconz-rest-plugin #2007: deCONZ crashes when opening hidden settings

peer69, 2019-10-26:
When trying to open the hidden gateway settings via Alt+click on "advanced" in the gateway settings, the deconz process goes to 100% CPU and crashes eventually. Is there another way to show these options? Trying to migrate to a Conbee II which seems to be impossible without.

peer69, 2019-10-26:
This is the latest log entry before deconz crashes: 

`void DeRestPluginPrivate::getZigbeeConfigDb(QVariantList&),4918: assertion 'rc == SQLITE_ROW' failed`

Smanar, 2019-10-26:
Yeah I think you have a problem, but hidden configuration is needed for a migration ?
If you use just backup and restore, it will not be enought ?

peer69, 2019-10-26:
Nope. Its a different GW. Restoring Phoscon config isnt enough in this case.

peer69, 2019-10-26:
A new install doenst show this problem. I guess my zll.db is just f*cked up. I'll do what I wanted to do for a long time now (due to connection issues), reset and repair about 65 devices and hope the systems runs smoother afterwards.

closed 2019-10-26 by peer69