dresden-elektronik/deconz-rest-plugin #2010: Make motion sensor work only at specific time range?

merdok, 2019-10-27:
Hello,

I would like to configure my motion sensor to only detect movement and turn on lights at a specific time in the Phoscon app. The only option which i found ar those: 
[image]

Is there something that i am missing? Is this even possible? Could somebody tell me how to do it?

Larsn1, 2019-10-29:
That is something that intrigues me as well.
I wish there would be a lot more options to choose from what happens and when.

merdok, 2019-10-29:
My sensor is located in the sleeping room, so i do not want the lights to go on when my wife is already sleeping there, so a specific time range is here a valid scenario. 

If this is not possible with the Phoscon app, does somebody know where the sources of the Phoscon app are located? I would at least try to modify that by myself and see if i get it to work...

sebastianlorenzen, 2019-10-29:
Connect a home automation system and do the logic there. And much more.

merdok, 2019-10-30:
But i would like to use Phoscon App for that. I do not want to install a dozen of home automation systems. I would like to have all configured in the Phoscon app.

So my question is, are the sources of the Phoscon app somewhere accessible?

ebaauw, 2019-10-30:
No, Phoscon is not open source.

You can use the REST API to create your own rules on the deCONZ server.  They’re powerful enough to do serious home automation, without the need for yet another home automation system.  Maintaining these is a bitch, though.  I use scripts for that, which also serve as a backup to re-create everything.

merdok, 2019-10-31:
The REST API would be good enough. Could you give me an example on how to set a time range for the sensor at which it should be active?

ebaauw, 2019-10-31:
That would be an additional condition in the rule, with an `in` operator for resource `/config/localtime` and value of something like `T07:00:00/T23:00:00`.  I actually only set a CLIP flag from the time, and use the flag in my motion sensor rules, so I can manually override the time (e.g. on new years eve).

The scripts I use to maintain my 250+ rules are in https://github.com/ebaauw/ph.sh.

merdok, 2019-10-31:
Thanks! Will try that out and see if i get it working through the REST API.

Do you know if it is also possible to like set the start time to sundown and end time to a specific time? Or does it always have to be a real time range?

ebaauw, 2019-10-31:
That would be two different rules, with different conditions: one on `/config/localtime` and one on `/sensors/1` (the internal daylight sensor).

The action takes a resource, `”/lights/1/state”`, a method, `”PUT”`, and a body, `{”on”: false}`, just like the regular API call.  Note that the value in a condition needs to be a string, whereas the value in the body is the actual type of the attribute.

merdok, 2019-11-01:
@ebaauw i get an error when trying to set the rule for address: `/config/localtime`
What am i doing wrong? 

```[
  {
    "error": {
      "address": "/config/localtime",
      "description": "Condition error",
      "type": 607
    }
  }
]

ebaauw, 2019-11-01:
You can only use operator `in` with `/config/localtime`, specifying a time range.  The condition evaluates to true during the entire range, but the rule fires only on the first time (assuming all other conditions are true).

merdok, 2019-11-01:
Yes, i have figured that out. I have configured a rule in the Phoscon app with active "from dusk till dawn" and i have added the time based condition from 7:30 to 22:00 to the rule afterwards. Let's see if that will work :)

Btw, when i take a backup from the gateway options in the Phoscon app. Will all this settings (even those which i set through the rest api) be backed up?

ebaauw, 2019-11-01:
Yes.  Phoscon is "just" another REST API client.  The settings don't know/care if they're set from Phoscon or from another client.

merdok, 2019-11-02:
Ok, everything works as expected after adding the localtime condition to the motion detection rule! Thank you!

closed 2019-11-02 by merdok