dresden-elektronik/deconz-rest-plugin #2011: [Device support request] Develco motion sensor MOSZB-130

jukmanty, 2019-10-27:
Unable to add this sensor via Phoscon app. It is not listed in the list of supported devices even though some Develco devices are supported.

https://www.develcoproducts.com/media/4314/moszb-130-140-technical-manual-motion-sensor.pdf

[image]

SwoopX, 2019-11-01:
You might wanna give it a try with the changes from my repo
[https://github.com/SwoopX/deconz-rest-plugin](https://github.com/SwoopX/deconz-rest-plugin)

You must compile the plugin with the given instructions.

jukmanty, 2019-11-02:
Thanks! I'll attempt to build it and try to pair the device.

jukmanty, 2019-11-06:
I was able to compile it to armhf architecture in raspbian/stretch docker container, but could not get it running in my RPi 3B+ + Raspbee + Hass.io environment. I modified official addon_core_deconz docker container https://github.com/home-assistant/hassio-addons/tree/master/deconz and copied compiled plugin to /usr/share/deCONZ/plugin/libde_rest_plugin.so Then I stopped and started container. Phoscon webapp starts, but REST API does not. Requests to /api are routed to webapp and it responds in html `<p>The file /api couldn't be found</p>`. I cannot login to Phoscon app, it does not accept any password. Also if VNC server port is configured then docker container exits with error code at VNC server startup.

If I add dbg-error=1 to deCONZ parameters I can see from the logs some errors during plugin lookup:

> 2019-11-06T18:53:20.963558000Z 20:53:17:688 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
2019-11-06T18:53:20.963868000Z 20:53:17:781 sub element size does not match real size
... 40 similar rows removed ...
2019-11-06T18:53:20.976747000Z 20:53:20:761 sub element size does not match real size
2019-11-06T18:53:20.977036000Z 20:53:20:764 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin

Can you assist how compiled plugin can be tested in my environment? Or is there something in the plugin code that prevents loading the library?

SwoopX, 2019-11-09:
Hm, I'm afraid I cannot be much of a help here, since I don't use docker. I do it the classic way with a dedicated test gateway and rpi and there it works ok. Maybe it has something to do with missing dependencies from the dev package.

You might want to open a separate issue for this. Maybe the other guys can help you with that specific behaviour.

SwoopX, 2019-11-30:
@jukmanty could you please post a screenshot from the node info pane for that device? Would help me to prove a theory...

jukmanty, 2019-11-30:
[image]

jukmanty, 2020-02-23:
I managed to add the sensor using latest beta version V2_05_73 built into hass.io deconz core add-on 5.2. Phoscon app reports that it found a sensor but it is not added to the list of sensors. But the sensor is visible in deconz GUI and in Home Assistant integrations as 4 motion sensors, 1 temp sensor and 1 battery level sensor. For some reason detected motion is visible in Home Assistant only after about 15 seconds (last binary sensor).

jukmanty, 2020-02-23:
[image]

SwoopX, 2020-02-23:
Hm, could you please provide the following screenshots (attributes must have been read): Occupancy sensing for ALL endpoints, illuminance measurement, IAS Zone and binary input?

According to the technical manual, you can potentially do pretty nifty things with it.

jukmanty, 2020-03-01:
Reading the attributes failed many times and succeeded very randomly (stating "reading done" in the dialog). Occupancy and illuminance measurement values changed automatically. Develco sensor was connected directly to the gateway. But here you go:

22 0406 Occupancy sensing

[image]

23 000F Binary Input

[image]

23 0500 IAS Zone

[image]

[image]

IAS Zone Settings could not be read.

27 0400 Illuminance measurement

[image]

28 0406 Occupancy sensing

[image]

29 0406 Occupancy sensing

[image]

jukmanty, 2020-03-01:
By the way Home Assistant started to report lux values at some point, currently using 2_05_74 (may be related deconz rest plugin or not)

[image]

SwoopX, 2020-03-01:
Thanks. I was primarily after the value of attribute 0x0011 for the occupancy regarding the delay. Looks ok to me. Also interesting to see that the Develco specific attributes are greyed out. Maybe you are running an older firmware and the attributes were introduced later on.

Is motion detection still with a delay or has it improved over time? I'd require your support to further investigate if required.

Regarding illumination, that might have taken a while. The sensor is currently configured to send a report every 10 minutes, whithout taking any change of illumination into account. You may change this behaviour via deconz GUI if desired.

jukmanty, 2020-03-01:
For grayed out values deconz GUI reported "unsupported attribute" after successful reading. I could try bind this device back to old Cozify hub which may be able to do the firmware OTA-update.

Delay is still there. I'm not also sure how to interprete these multiple occupancy sensing values, but it seems that one reacts to motion quite reliably but with delay. Two other react for example if you move the sensor. I'm happy to help to the best of my ability.

SwoopX, 2020-03-01:
> For grayed out values deconz GUI reported "unsupported attribute" after successful reading. I could try bind this device back to old Cozify hub which may be able to do the firmware OTA-update.

For integration, I took the latest technical manual available. It was refering to 3 manufacturer specific attributes... If you want to do that to satisfy my curiosity, why not. Shouldn't cause any harm. Btw, I envy you for the hub ;)

Currently, my theory for the delay is that the binding table is full and therefore, reporting is not active, but I might be wrong. E.g. I'vee seen that the innr SP120 allow for 6 bindings, but your sensor has more potential binding targets. We should double check that via deconz GUI, do you know how to do that?

jukmanty, 2020-03-02:
Nope... can you guide me or give link to documentation?

SwoopX, 2020-03-03:
Take a look at here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/812#issuecomment-425660327

For the source part, just take the occupancy sensors and clusters. You can configure the attribute reporting for occupancy (0x0000) then. You may try something like values 1, 60, 0

Could you also please provide the REST API output for the sensors created?

jukmanty, 2020-05-24:
Sorry about the delay.

Here are binding box screenshots:

[image]
[image]
[image]

I could not change 0x0000 values of occupancy clusters. Those are read-only.

REST API sensors extract:

    ...snip...

    "19": {
        "config": {
            "duration": 5,
            "on": true,
            "reachable": true
        },
        "ep": 34,
        "etag": "f9312583582208d25fc70bc7c4a5396b",
        "lastseen": "2020-05-24T20:20:20.037",
        "manufacturername": "Develco Products A/S",
        "modelid": "MOSZB-130",
        "name": "Motion Sensor (2)",
        "state": {
            "lastupdated": "2020-05-24T20:19:16.990",
            "presence": true
        },
        "swversion": "2017-09-12 01:19",
        "type": "ZHAPresence",
        "uniqueid": "00:15:bc:00:1a:00:76:b0-22-0406"
    },
    "20": {
        "config": {
            "battery": 90,
            "duration": 5,
            "on": true,
            "pending": [],
            "reachable": true
        },
        "ep": 35,
        "etag": "1064c163321b13983c85950b9614a2bb",
        "lastseen": "2020-05-24T20:20:20.037",
        "manufacturername": "Develco Products A/S",
        "modelid": "MOSZB-130",
        "name": "Motion Sensor (2)",
        "state": {
            "lastupdated": "2020-05-24T20:18:53.121",
            "lowbattery": false,
            "presence": false,
            "tampered": true
        },
        "swversion": "2017-09-12 01:19",
        "type": "ZHAPresence",
        "uniqueid": "00:15:bc:00:1a:00:76:b0-23-0500"
    },
    "21": {
        "config": {
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 38,
        "etag": "1028dd2ded4da781555dea9ab07b5caa",
        "lastseen": "2020-05-24T20:20:20.042",
        "manufacturername": "Develco Products A/S",
        "modelid": "MOSZB-130",
        "name": "Motion Sensor (2)",
        "state": {
            "lastupdated": "2020-05-24T20:20:20.037",
            "temperature": 2287
        },
        "swversion": "2017-09-12 01:19",
        "type": "ZHATemperature",
        "uniqueid": "00:15:bc:00:1a:00:76:b0-26-0402"
    },
    "22": {
        "config": {
            "on": true,
            "reachable": true,
            "tholddark": 12000,
            "tholdoffset": 7000
        },
        "ep": 39,
        "etag": "1c080727fb248e3b91995fbff476dd55",
        "lastseen": "2020-05-24T20:20:20.037",
        "manufacturername": "Develco Products A/S",
        "modelid": "MOSZB-130",
        "name": "Motion Sensor (2)",
        "state": {
            "dark": true,
            "daylight": false,
            "lastupdated": "2020-05-24T20:16:03.000",
            "lightlevel": 9625,
            "lux": 9
        },
        "swversion": "2017-09-12 01:19",
        "type": "ZHALightLevel",
        "uniqueid": "00:15:bc:00:1a:00:76:b0-27-0400"
    },
    "23": {
        "config": {
            "duration": 5,
            "on": true,
            "reachable": true
        },
        "ep": 40,
        "etag": "7c8f7a1720022f0182fa6983347d762c",
        "lastseen": "2020-05-24T20:20:20.037",
        "manufacturername": "Develco Products A/S",
        "modelid": "MOSZB-130",
        "name": "Motion Sensor (2)",
        "state": {
            "lastupdated": "2020-05-24T19:15:23.120",
            "presence": false
        },
        "swversion": "2017-09-12 01:19",
        "type": "ZHAPresence",
        "uniqueid": "00:15:bc:00:1a:00:76:b0-28-0406"
    },
    "24": {
        "config": {
            "duration": 5,
            "on": true,
            "reachable": true
        },
        "ep": 41,
        "etag": "2ec308279198fbc87b8dcddc0bcbad38",
        "lastseen": "2020-05-24T20:20:20.037",
        "manufacturername": "Develco Products A/S",
        "modelid": "MOSZB-130",
        "name": "Motion Sensor (2)",
        "state": {
            "lastupdated": "2020-05-24T20:19:22.018",
            "presence": false
        },
        "swversion": "2017-09-12 01:19",
        "type": "ZHAPresence",
        "uniqueid": "00:15:bc:00:1a:00:76:b0-29-0406"
    },

Home assistant deCONZ integration device page:

[image]

Currently I'm using Home Assistant 0.110.1 with deCONZ plugin 5.3.3 = deCONZ release V2_05_76 - Howlin’ Wolf. Presence state changes are still very uncertain. Last one seems to react most reliably, but with 10-20 sec delay and not 100% accurately. Sometimes first and second react very quickly to motion, but then values gets stuck into "Detected" or "Clear" states.

I was not able to upgrade firmware via Cozify hub. I cannot control it anymore in any way as I have cancelled the monthly subscription (so nothing to envy about that... it's practically electronic garbage).

SwoopX, 2020-05-24:
It's your device, not mine ;)

Great that you run .76 already since there were some improvement. I need some more info however.
Please double click on the IAS zone cluster, scroll down to the bottom. You should see 2 Develco specific attributes. Please double click on each of them and read the attribute (not "read config"). `Alarm Off Delay` should be the period in seconds for the sensors jumping from "detected" to "Clear". I'd assume sensor with ID 20 is working best/adequately...?

jukmanty, 2020-05-25:
I could not read the attribute, tried several times:

[image]

And yes, ID 20 sensor reacts most reliably, but has the delay. Others may react to motion more quickly, but state tends to get stuck either to "Clear" or "Detected"

SwoopX, 2020-05-25:
Hm, it seems that the 2 attributes have been introduced just latetly and maybe require a firmware update to become available. So based on the information provided, I'd expect that motion is pretty much instandly detected by sensor 20, but it may take some time to fall back to clear (I see duration is 5 seconds). Now if within that 5 second motion is detected again, the remaining time should be added another 5 seconds and so forth. You could try setting config duration via REST API to 0 and see if that better fits your needs.

Also, for the remaining sensors, the device as quite some complexity to offer. I'd recommend to check the technical manual again, since you can steer the occupancy setting/detection based on multiple additional values.

Maybe I can also spot anything in the code which might add any additional delay here.

closed 2020-07-04 by stale[bot]

WJ4IoT, 2022-02-24:
@SwoopX thanks for pointing me towards this location. I followed the instruction '**Precompiled deCONZ packages for manual installation**' for two reasons I got stuck on the installment of qmake and this way looked more the dummies/NOOB-way. I am using a Pi4 headless with installed Openhabian and Conbee II,

In the end I did not see much difference for the MOSZB-140 in the Phoscon APP compared to the official version; The **version** is now unknown and the Lightlevel gives (still) a bit out of ranges values like 21085 LUX. Perhaps the LUX value is not available (*) because according to the specification  of the light sensor there are only the resolution values: dark, light, bright available ([source](https://frient.com/products/motion-sensor-pro/)). So not sure if I may expect correct LUX values.

(* = I see for example in another sensor (Mi-Light Detection) a constant temperature value were there is no temperature sensor included)

I still need further examine what is available in Openhabian

SwoopX, 2022-02-26:
Fetching the version is kinda irrelevant and since the devices are deep sleepers, that's more a hit and miss. This works way better with the new DDF code and with the respective DDF about to be included, it works way more reliable and gives you this time the true device firmware version (which can currently only been seen in a sniffer log).

The wonky values is what the device reports, deconz doesn't really manipulate that. However, lux values are calculated from the reported lightlevel readings. And then it's sh*t in, sh*t out. Interestingly, I just checked one of my sensors and it seems to also report garbage:

[image]

That would mean I had bright daylight in the middle of the nicht in my kitchen, which I guarantee I had not :) Hope to catch this any time soon in a sniffer log to see what's going on.
I got 2 sensors actively in use, one from each "brand". Gotta check which one it is but I know they have both different firmware versions...

One remark regarding your source: that is a bit nonsense. Frient, in fact, is Develco (same company address, same devices and hardware). Develco has great technical manuals and those document what the device is capable of.