dresden-elektronik/deconz-rest-plugin #2013: Integrate SwannOne BX_SWO-OUS1P

xhemp, 2019-10-28:
Trying to get a BX_SWO-OUS1P  SwannOne Siren on my deCONZ.
Thing is that the siren is being recognised as light on deCONZ although it is showing up on HomeAssistant as a switch.
Even being a switch on HA, I can switch on the siren (so I would be able to have it ringing when the alarm is triggered) but I am not able to switch it off.
Here are the screenshots i got from deCONZ:

[image]
[image]
[image]
[image]
[image]

xhemp, 2019-10-28:
Looks like this has been looked at #404.

xhemp, 2019-11-03:
I was able to use the state.alert to control the siren (none/select/lselect work as intended).
The only thing that I couldn't get it working was "blink", it does the siren and the strobe at the same time.

xhemp, 2019-11-18:
Re-opening this one as I would like to get this sorted. The siren work fine for me (even though the strobe doesn't work on state.alert = blink but it's not a big deal.
Problem with it is that it just loses connection after some hours and I can't control the siren anymore. It also beeps sometimes out of the blue with no interaction at all. Just wondering if something else needs to be done on its configuration.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix