dresden-elektronik/deconz-rest-plugin #2014: Devices randomly lose connection

sweebee, 2019-10-28:
deconz version: 2.05.69
firmware: 26330500
gateway: raspbee

Recently since a few weeks devices randomly lose connection to the gateway. Like today it lost the connection to my hall light upstairs (Hue white bulb) and the led strip light in the kitchen (Ikea 24 dimmer).

I disconnected them from the power for a few minutes and reconnected them. sometimes they work for a few minutes and sometimes they wont do anything at all.

few days back i managed to solve it by let it be disconnected (from the power) for a day and reconnected them again to the power and they worked again.

I cant remember that anything changed in the past weeks. Also when my raspberry loses power its a pain in the ass to get every light working again.

I have 29 lights and 10 sensors connected. Maybe this is too much?

sweebee, 2019-10-28:
Kitchen light now working again.

Also a weird thing, the light upstairs now always says its connected, but it wont dim or turn on/off

sweebee, 2019-10-28:
Some log info when trying to turn on/off the light upstairs:

```
19:48:57:335 ZCL attribute report 0x001788010646DDE9 for cluster 0x0402, ep 0x02
19:48:58:842 Current channel 25
19:48:58:864 Device TTL 3900 s flags: 0x7
19:48:59:746 no button map for: SML001 ep: 0x02 cl: 0x0402 cmd: 0x0A pl[0]: 000
19:48:59:746 ZCL attribute report 0x0017880104B7B78F for cluster 0x0402, ep 0x02
19:49:01:798 trigger rule 43 - pir-fsm-state-end
19:49:01:801 rule event: /sensors/40 state/status num (3 -> 0)
19:49:01:814 rule event: /groups/10 state/any_on num (0 -> 10)
19:49:02:584 0x0017880110529F21 error APSDE-DATA.confirm: 0xA7 on task
19:49:05:808 max transmit errors for node 0x00158D0001F43189, last seen by neighbors 1215 s
19:49:06:954 0x0017880110529F21 error APSDE-DATA.confirm: 0xA7 on task
19:49:08:765 remove outdated neighbor 0x5E47
19:49:13:765 remove outdated neighbor 0x067F
19:49:15:942 delay sending request 52 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:15:967 delay sending request 52 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:16:065 delay sending request 52 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:16:166 delay sending request 52 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:16:266 delay sending request 52 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:16:365 delay sending request 52 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:16:414 delay sending request 52 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:16:465 delay sending request 52 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:16:566 delay sending request 52 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:16:665 delay sending request 52 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:16:767 delay sending request 52 dt 4 ms to 0x0017880110529F21, cluster 0x0006
19:49:16:866 delay sending request 52 dt 4 ms to 0x0017880110529F21, cluster 0x0006
19:49:16:966 delay sending request 52 dt 4 ms to 0x0017880110529F21, cluster 0x0006
19:49:17:065 delay sending request 52 dt 4 ms to 0x0017880110529F21, cluster 0x0006
19:49:17:166 delay sending request 52 dt 4 ms to 0x0017880110529F21, cluster 0x0006
19:49:17:265 delay sending request 52 dt 4 ms to 0x0017880110529F21, cluster 0x0006
19:49:17:369 delay sending request 52 dt 4 ms to 0x0017880110529F21, cluster 0x0006
19:49:17:465 delay sending request 52 dt 4 ms to 0x0017880110529F21, cluster 0x0006
19:49:17:517 delay sending request 52 dt 4 ms to 0x0017880110529F21, cluster 0x0006
19:49:17:565 delay sending request 52 dt 4 ms to 0x0017880110529F21, cluster 0x0006
19:49:17:665 delay sending request 52 dt 4 ms to 0x0017880110529F21, cluster 0x0006
19:49:23:765 remove outdated neighbor 0x2D29
19:49:24:798 0x0017880110529F21 error APSDE-DATA.confirm: 0xA7 on task
19:49:28:850 0x0017880110529F21 error APSDE-DATA.confirm: 0xA7 on task
19:49:31:448 delay sending request 225 dt 1 ms to 0x0017880110529F21, cluster 0x0006
19:49:31:473 delay sending request 225 dt 1 ms to 0x0017880110529F21, cluster 0x0006
19:49:31:565 delay sending request 225 dt 1 ms to 0x0017880110529F21, cluster 0x0006
19:49:31:666 delay sending request 225 dt 1 ms to 0x0017880110529F21, cluster 0x0006
19:49:31:767 delay sending request 225 dt 2 ms to 0x0017880110529F21, cluster 0x0006
19:49:31:866 delay sending request 225 dt 2 ms to 0x0017880110529F21, cluster 0x0006
19:49:31:884 Replace task 2858215 type 14 in queue cluster 0x0006 with newer task of same type. 1 runnig tasks
19:49:31:885 delay sending request 231 dt 2 ms to 0x0017880110529F21, cluster 0x0006
19:49:31:966 delay sending request 231 dt 2 ms to 0x0017880110529F21, cluster 0x0006
19:49:32:065 delay sending request 231 dt 2 ms to 0x0017880110529F21, cluster 0x0006
19:49:32:165 delay sending request 231 dt 2 ms to 0x0017880110529F21, cluster 0x0006
19:49:32:268 delay sending request 231 dt 2 ms to 0x0017880110529F21, cluster 0x0006
19:49:32:366 delay sending request 231 dt 2 ms to 0x0017880110529F21, cluster 0x0006
19:49:32:466 delay sending request 231 dt 2 ms to 0x0017880110529F21, cluster 0x0006
19:49:32:490 Replace task 2858219 type 14 in queue cluster 0x0006 with newer task of same type. 1 runnig tasks
19:49:32:490 delay sending request 238 dt 2 ms to 0x0017880110529F21, cluster 0x0006
19:49:32:565 delay sending request 238 dt 2 ms to 0x0017880110529F21, cluster 0x0006
19:49:32:665 delay sending request 238 dt 2 ms to 0x0017880110529F21, cluster 0x0006
19:49:32:766 delay sending request 238 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:32:865 delay sending request 238 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:32:965 delay sending request 238 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:33:066 delay sending request 238 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:33:165 delay sending request 238 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:33:266 delay sending request 238 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:33:366 delay sending request 238 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:33:465 delay sending request 238 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:33:565 delay sending request 238 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:33:665 delay sending request 238 dt 3 ms to 0x0017880110529F21, cluster 0x0006
19:49:33:766 delay sending request 238 dt 4 ms to 0x0017880110529F21, cluster 0x0006
19:49:33:865 delay sending request 238 dt 4 ms to 0x0017880110529F21, cluster 0x0006
19:49:33:966 delay sending request 238 dt 4 ms to 0x0017880110529F21, cluster 0x0006
```

greencharlie100, 2019-10-30:
I am having similar issues with devices (switches and sensors) losing connection. There does not seem to be support for deconz anymore!

andriej, 2019-10-30:
Same issue lately - xiaomi movement and door sensors loose some of the state changes and come back to work unexpectedly. Can't debug right now what is the issie, especially when it's only part of them.

zigistar, 2019-10-30:
same here - sometimes my Philips Hue lights, my Heiman smoke sensors and my Aqara sensors loose connection. That happens absolutely randomly to the lights and sensors. Sometimes, it I can reconnect them by reseting the device in Deconz (Edit-Menu + "reset selected node F5"), sometimes I have to select "Reboot device" too. Sometimes I have to add/pair the devices again in Phoscon to make them work again, especially for the Heiman smoke sensors this is the only way. This happens not regularly but from time to time, I'd say it happens once in a week for one light/device/sensor. Rebooting the Raspi is not helping for Heiman smoke sensor - just had this issue today with the smoke sensor, reboot hasn't changed anything, re-pairing was necessary to get the sensor connected again.

I'm running 2.05.69 and firmware 26330500 on a Raspi 3+

incmve, 2019-11-13:
I got the same problems, after a few hours no data comes in.
Yesterday I installed a Pi3 with version 02.05.**67** and firmware 264A0700 and now (24 hours later)  it still works, alls sensors are reporting data.

greencharlie100, 2019-11-14:
My heating did not switch on this morning again. Seems that the network is very unreliable.

Here is a common error that shows up in Home Assistant logs.

> Sun Nov 10 2019 02:39:21 GMT+0000 (Greenwich Mean Time)

Error doing job: Fatal read error on socket transport
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/selector_events.py", line 723, in _read_ready
    data = self._sock.recv(self.max_size)
TimeoutError: [Errno 110] Connection timed out
> 

@manup Can you help resolve this, please.

greencharlie100, 2019-11-14:
Also, my temperature sensors seem to report data for hours and then go invisible for a long time and then randomly start reporting data again.

PeterRemnemark, 2019-12-11:
I've similar error, where some IKEA tradfri bulbs does not react on the commands from Conbee. In both UI's (Phoscon and Home Assistant) the device looks to be online and in the suppose state, but when physical it is not in same state when the error show up. If you try to change the status of the Bulb there is no action delivered to the actual bulb. 
The randomness is with a few selected devices. If I let the device just let them be as they are, they would "heal" over a period of time. I.e. if I see one bulb being in on state over the night (when it suppose to be off), the bulb would be acting on commands when the morning comes and after sunrise the bulb now turns off. I cannot really see what the error resides, I've changed the channel to 25 and made sure the WiFi is not interfering either (no close neighbors to interfere), and it got a bit better but not completely solved). This has been working quite well for a long time, and it started after I added several new bulbs (approx. 25 new), but perhaps not direct after. 

Anyone have a clue to trace or check?

fwestenberg, 2020-01-12:
Same here, with Xiaomi sensors and Ikea lights. Very random which devices is losing connection. My only clue is the dead link container (size 211)...

toellbe, 2020-04-22:
Same here. ICZB-IW11D Lost Connection / Is grayed out. Reboot Gateway solve the issue for minutes - max hours. 

Xiaomi Sensors not affected.

madiskallas, 2020-05-07:
Is this issue still not resolved? The whole system is unusable like this.

incmve, 2020-05-07:
I think it is, but I am running 5 months without issues with the older version.
02.05.67 and conbee II firmware 264A0700
Got Ikea lights, plugs and Xiaomi sensors around the house.

madiskallas, 2020-05-07:
I have the latest version (2.05.75 and 26350500) and the performance is really bad. I am loosing lights/switches/sensors daily, different entities every day. I have to repair them to use again.

Smanar, 2020-05-07:
Hello,
@madiskallas if you loose your device fast enought, can you make a try on another machine to try ?
like https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2710

madiskallas, 2020-05-08:
I'll give it a try, yes. Thank you.

fwestenberg, 2020-05-08:
I also struggled with this issue, having a lot of light bulbs as repeater I didn't think kf connectivity issues. But since I moved the gateway to the center of my house, everything is stable now!

madiskallas, 2020-05-08:
My gateway is almost in the center of all the devices (about 70 devices). But also devices 2 meters from the gateway loose connection. The problem must be somewhere else. I will now reinstall deConz from scrach and see how it works out.

Smanar, 2020-05-08:
@fwestenberg 
You have moved only the conbee, not the machine.
You have used an USB cable extension (and not used before) ?
There is nothing special in your home that can explain that ? Router, microwave oven, glass wall, .....

fwestenberg, 2020-05-08:
I already had an extender cable in use, but it didn't help me much. My machine was connected by ethernet, so WiFi interference didn't play parts I guess. And I indeed moved the whole machine to a more central location (1st floor). It's really amazing but all my connection problems are solved for months now.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

hwinkel, 2020-07-01:
Hi, we have exactky the same problem with at least two gateways updated to version. 75. The whole system is unusable now. And imho the issue can't be resolved as it just contains failure reports of the user but not really a single idea how to resolve. The issue for us appeared after the update.

Smanar, 2020-07-01:
Which one router / end device band have you ?
Have you a capture of your mesh ?
Are you using USB extension cable ?

Mimiix, 2020-07-01:
@hwinkel  Can you open a new [Bug report](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=Bug+report&template=bug_report.md&title=) ?