dresden-elektronik/deconz-rest-plugin #2015: Restarting deconz computer fails connecting nodes

TheWizz, 2019-10-29:
I have a simple test system in preparation for a larger system to be deployed shortly. It currently consists of a single HUE light and an Innr switch. I can add those devices OK to deconz, and they operate normally. I can quit and relaunch deconz, and they appear and re-connect OK. Hoerver, if I quit deconz, disconnect the ConBee dongle, plug in the ConBee Dongle then restart deconz, they fail to reconnect. 

[image]

They first appear to connect (green wires appear), but the light and the INNR plug flash red instea of the normal blue circle. After a while, the green wires disappear.

I can make both devices reconnect again by turning power to the HUE light off then on again. Then they always reconnect, and work normally.

[image]

I can also recreate the same problem by restarting the Ubuntu box I run this on. After restart, the same problem as described above happens. Again, I can fix it by power-cycling the HUE light.

I've tested both the current shipping version of deconz as well as the current beta. Both behave the same. This is running on Ubuntu 16.04, starting deconz in GUI mode as a regular program.

-JM

JensTimmerman, 2019-11-03:
A lot of people seem to have this issue, including me, for the past month now, I would like to help fix this, but I can't find any information on where to start debugging this.

Mimiix, 2020-06-10:
As per request re-opened. Can you add some details on versions and such?  What firmware version are you running and what version of deconz is running?

TheWizz, 2020-06-10:
Thanks for re-opening. Since the original posting, we've made some further tests. We have this problem ONLY if all the devices are Philips HUE lights. As soon as we add another device (such as an Innr plug) to the network, that seems to "fix" the problem. All devices then appear after a reboot. But if we remove the Innr plug, leaving only the HUE lights in the system, the problem comes back. That may explain why not everyone has this problem, as many users have a mixed set of devices.

The HUE lights are all brand new. When they're not seen by ConBee, they can still be turned on/off using the Hue remote, so the zigbee bus seems to work fine.

-JM

Mimiix, 2020-06-10:
Also add the Device type (conbee 1 or 2?) and logs if that happens. 

Also, what is the host device?

TheWizz, 2020-06-10:
It was a Conbee I, and we were running it on Ubuntu 16.04 on an Intel NUC. Regarding version numbers, I will need to take a look since this was quite a while back.

-JM

Mimiix, 2020-06-10:
Sounds good. Just update all to latest first, then check again.

Also: you are running it on bare nuc? No VM involved?

Mimiix, 2020-06-26:
@TheWizz Hey there! How's things? Any news?

TheWizz, 2020-06-26:
I'm sorry. Summer holiday got in the way. We'll pick this up after the break. Hopefully, by then the updated firmware/software has been released, so we can just install that and see if it works OK now. Will let you know.

-JM

Mimiix, 2020-06-26:
I think the update is already out ;)

For now, i'll close this issue then. Tag me when your back. Enjoy your summer! #familyfirst :)

closed 2020-06-26 by Mimiix