dresden-elektronik/deconz-rest-plugin #2016: RaspBee lost connection

halloamt, 2019-10-30:
Hi,

yesterday my RaspBee lost its connection to my ZigBee network. deConz shows all the nodes but without connection lines.
What I am using ist a RaspBee together with a RPi 4B 4G and Raspbian Buster. DeConz is 2.05.70. I tried manually installing 2.05.69 and flashing the older firmware. Flashing works perfectly, but I could not get a connection to my ZigBee network.
What happened is that my RPi got a new EEPROM image (000137ab) rebooted the Pi and tested everything. For a few hours it worked and then suddenly stopped working after a few hours.
Now my RaspBee's red LED is on and the green one approximately blinks once in 10 or 20 seconds. DeConz has a connection to my RaspBee, it shows all devices and the correct FW version. If deConz is running, I also cannot flash any firmware.
Could that be a hardware problem?

Greetings from the Black Forest

manup, 2019-10-30:
>Now my RaspBee's red LED is on and the green one approximately blinks once in 10 or 20 seconds.

That's normal operation mode.

>What happened is that my RPi got a new EEPROM image (000137ab) rebooted the Pi and tested everything. 

Is that a new operating system update? 

>DeConz has a connection to my RaspBee, it shows all devices and the correct FW version. If deConz is running, I also cannot flash any firmware.
Could that be a hardware problem?

Does deCONZ show that the network is connected as "In Network"?

[image]

I'd also recommend you check the Zigbee configuration page in the Phoscon App as described in: https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

On this page you can see if some network parameters have been changed recently and revert back if that's the case.

halloamt, 2019-10-30:
That is the new Boot-EEPROM. No OS updates involved. See https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md and https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/release-notes.md
deCONZ shows as "In Network", neither the PID nor the key have changed. I do not remember wheter the Update ID changed.
I'll have a fresh deCONZ installation ready and can test that later at home.

manup, 2019-10-30:
>That is the new Boot-EEPROM. No OS updates involved. See https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md and https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/release-notes.md

Cool very interesting, I didn't know about that.

>deCONZ shows as "In Network", neither the PID nor the key have changed. I do not remember wheter the Update ID changed.
I'll have a fresh deCONZ installation ready and can test that later at home.

Ok, strange can you also try to power-cycle one of the non-responding lights to check if it comes back as described in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2015

halloamt, 2019-10-30:
Powercycling a lamp did not work, I tried that yesterday. But after a restore of an old backup everything worked again. I can give you the working and the not working state as backup if that helps...

manup, 2019-10-31:
>P.S.: None of the lights and none of the sensors were connected. It seems my RaspBee changed its channel or something like that.

A channel change would be visible in the Zigbee configuration page (and the backups as well).

>P.P.S.: Just that I do not forget tomorrow. I sent the backups to info@dresden-elektronik.de

Thanks we'll check them if there is anything suspicious.

closed 2020-03-06 by stale[bot]