dresden-elektronik/deconz-rest-plugin #2017: Idea for an extension: Switch editor Time of day condition

Nastras, 2019-10-30:
Hello,

wouldn't it be an idea in the editor switch to allow scenes to be switched depending on time of day?

As an example I want the light to always be set to 100% from sunrise to sunset. 

And from sunset to sunrise the light should be switched on only 50%. 

If that would be possible it would offer some new possibilities to program switches in Phoscon.

What do you think is this possible? @manup

Krocko, 2019-10-30:
Hi @Nastras 
You can do this easily with the rest api and the build in daylight sensor.

Nastras, 2019-10-30:
Really, I didn't know that. I would have expected this function in phoscon. For the simple user.

Where exactly can I find it in the rest api?

Krocko, 2019-10-30:
Just create rules with an rest api client. 
See the rest api documentation http://dresden-elektronik.github.io/deconz-rest-doc/
You can use the Phoscon switch editor, to assign the buttons. Then add the condition for day with the rest api client. Copy this rules, change the brightness and the condition to night.

Nastras, 2019-10-30:
Cool. Learned something again. I'll take a closer look later. Thank you! Of course it would be nice to be able to set it in Phoscon via the web interface.

closed 2019-11-06 by Nastras