dresden-elektronik/deconz-rest-plugin #2020: Segmentation fault when launching deCONZ program

SamSamFR, 2019-10-30:
After upgrading from 2.05.67-raspbian-stretch-beta to 2.05.70-raspbian-stretch-beta (along with other raspbian packages, see log attached
[aptitude.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3791079/aptitude.log)
), I systematically get segmentation fault a couple seconds after program launch when running : 
- deCONZ -platform minimal --http-port=8080
- deCONZ  --http-port=8080

BUT in the latter case, if I move the mouse to select ConBee II and click connect, the app does work stable afterwards. 
This bug seems like it's probably related to a X or kde library dependency since user interaction can avoid the bug.

What do you need me to do to help diagnose the problem ?

manup, 2019-10-30:
Most helpful would be a coredump to track down where the SEGV happens.
Can you please create one as described in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Enable-core-dumps-on-Raspbian

SamSamFR, 2019-10-30:
Sure, here's the core dump, from running deCONZ -platform minimal --http-port=8080

[core-deCONZ-sig11-user1000-group1000-pid2631-time1572476196.bz2](https://drive.google.com/open?id=18Jwj6N0mlWnlSde1Hj7E8Kj3q0MIIrqs)

Empha88, 2019-11-07:
I also got this problem. 
After reboot it starts the GUI, connects to Conbee, you see the devices and then the GUI shuts down and the process starts all over again and again and again...

manup, 2019-11-07:
Can you please run deCONZ while debug log is enabled.

    deCONZ --http-port=8080 --dbg-info=1 --dbg-error=1

Empha88, 2019-11-07:
[deconz.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3818674/deconz.txt)

Empha88, 2019-11-07:
> Can you please run deCONZ while debug log is enabled.
> 
> ```
> deCONZ --http-port=8080 --dbg-info=1 --dbg-error=1
> ```

[deconz.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3818686/deconz.txt)

manup, 2019-11-07:
Thanks, looks like this is related to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2033

My commit https://github.com/dresden-elektronik/deconz-rest-plugin/commit/59b7a04abf2c78f3d7c1e2c2e2f8984e06fe1022 seems to cause the issue. By now I think after the reconnect the app is restarted which was ok in another scenario but causes issues here. ... working on a fix.

Empha88, 2019-11-07:
> Thanks, looks like this is related to #2033
> 
> My commit [59b7a04](https://github.com/dresden-elektronik/deconz-rest-plugin/commit/59b7a04abf2c78f3d7c1e2c2e2f8984e06fe1022) seems to cause the issue. By now I think after the reconnect the app is restarted which was ok in another scenario but causes issues here. ... working on a fix.

Appreciate the help, hope you find a solution, thanks!

closed 2020-03-13 by stale[bot]