dresden-elektronik/deconz-rest-plugin #2021: Trouble discovering sengled bulbs

Leslik, 2019-10-30:
I just bought a Conbee2 and trying to discover some Sengled bulbs and not getting very far. I can see others have managed it but I can't seem to find them as end devices.

I can't find them in the Phoscon webapp and this is all I can seem to find with deconz

[image]

Sorry if the question is very basic but since this is my first go at any kind of home automation i'm still on the starting blocks. 

My guess is that all I'm finding is the mesh part of the bulbs talking to each other but none of the capabilities of the bulbs themselves. I've done a lot of searching and google is not being my friend

Any ideas what I am doing wrong?

ebaauw, 2019-10-31:
deCONZ hasn’t received the Simple Descriptors.  You can try and have deCONZ request them again through the option in the dropdown list.  The status “light” on the left should blink blue when the ConBee sends or receives messages from the device.  And red when the response times out.

Leslik, 2019-10-31:
Ok thanks, that helps. The lights are blinking red for timeout. If I click the simple descriptor button it turns straight back off and won't stay on. Is that normal?

I see blue flashing light on the conbee. The bulbs do a short green flash when first detected. Sometimes a long yellow flash followed by a 3 red flashes a few seconds later. This is after I power the bulbs off and on 10 times to reset them. The last contact time updates for a minute or two and then stops.

This is what I get for node info

[image]

closed 2020-03-06 by stale[bot]