dresden-elektronik/deconz-rest-plugin #2023: I have to re-pair all my devices after a Deconz restart

mihalski, 2019-11-01:
I've only had the Conbee II for a few days but I've already had to re-pair 42 devices 4 times over. After any restart almost all of them stop functioning via home assistant.

On the network map even if they appear and show as connected and the blue indicator flashes when you press a button, this does not get passed to home assistant and rarely even shows up in the Deconz logs.

Please let me know what info could help debug and resolve this.

Smanar, 2019-11-01:
>the blue indicator flashes when you press a button

You can send command too ? for exemple to turn on a bulb ?
You have the green lines in the network map ?
It works trought Phoscon ?
Have you try the power cycle ?

And you are not alone with this bug > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2015

FelixUster, 2019-11-05:
Maybe a stupid question, but did you try to restart Home Assistant? If I only restart deconz the connection to deconz is somewhat broken, too. I also need to restart home assistant

pergolafabio, 2019-11-05:
is this maybe related? that "was" my issue

https://github.com/home-assistant/home-assistant/issues/27531

stoneobscurity, 2019-11-07:
are you running deconz in a docker?

closed 2020-03-13 by stale[bot]