dresden-elektronik/deconz-rest-plugin #2024: Reinstall, Phoscon identifies wrong device

surfarn, 2019-11-01:
My PI died on me, so I moved my Conbee to a Windows laptop to add all the devices again. 

But now it does not identify all products correctly.

Here you can see that I added two Aqara motion sensors, one of the was identified as Ikea motion sensor.
[image]

I also had problems to add my 4th aqara switch, the first 3 were identified correctly, but when I added the third, it added an Ikea Trådfri remote controll and two lights.

[image]
 
[image]

I do have two Trådfri Remote controlls, but I have not put them in pairing mode yet. 

Why does this happen? Should I remove the devices in Phoscon and pair them again? I am using 2.05.69 and firmware 26330500

Smanar, 2019-11-01:
Hello, It s a 100% fresh install on windows, or have you do a backup/restore ?

surfarn, 2019-11-01:
It is a 100% fresh install, lost the backups when the PI died as well.

ebaauw, 2019-11-01:
Might be radio interference.  How did you connect the ConBee to the laptop?  Do you use an USB extension cable?  Is it plugged into a USB2 port?  Are WiFi and Bluetooth on the laptop disabled?  Does the laptop have USB3 devices connected?

surfarn, 2019-11-01:
I have connected it directly to an USB port on the laptop, both WiFi and Bluetooth are activated. This is the only USB device that is connected, don't know if it was the usb3 or usb2 that I used, will have to check when I get back home. 

I have an USB extenstion cabel at home, could try to use it to see if that makes any differens. 

Can I always remove devices from the Phoscon app and try to pair them again? Or do I need to find a way to delete them from the database first?

ebaauw, 2019-11-01:
Normally it's enough simply to reset the device and re-pair it.  deCONZ should fix any stale node info.  If the REST API plugin created really weird resources, best delete those using the API (with a `DELETE` of `/api/`_apikey_`/lights/`_id_ or `/api/`_apikey_`/sensors/`_id_) after resetting the device.

Note that if the issue is caused by poor radio reception, resetting and re-pairing the device will likely run into the same problems.  I might be more productive to try and manually re-start the pairing process where it stalled.   See https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2002#issuecomment-547985838.

surfarn, 2019-11-02:
I managed to get it to work! 

I did a complete gateway reset. Changed so that a usb2 port was used instead. I also added an extention-cord on the Conbee. 

I have now managed to pair all the devices correctly.  Will now do a backup and move the Conbee to my server, instead of the lapptop. 

Thanks for the help!

closed 2019-11-02 by surfarn