dresden-elektronik/deconz-rest-plugin #2026: Phoscon Web UI not compatible with lots of devices

cputoaster, 2019-11-01:
The phoscon web UI has some logical problem with many lights:

- if there are a lot of lights (around 100 for me), a normal notebook screen is too small to show them for adding to a group. A simple fix would be to add scrolling to the light panel.
- if a new light is added, it is added at the bottom of the current lights list, which is invisible without manually scrolling. There is no visual cue if the light was detected or not, one always has to try to scroll to see if it worked. Ideally there would either be a pop-up, the list would automatically scroll up, or the new light be added on top so its visible.

manup, 2019-11-01:
Thanks for the report, I thought this was fixes already, we'll check the code.

manup, 2019-11-19:
The issue is fixed now in 2.05.71.

closed 2019-11-19 by manup