dresden-elektronik/deconz-rest-plugin #2027: Strange behaviour

ademobile, 2019-11-01:
I have two devices that are connected to another system, yet they are also shown as connected to deCONZ too! I have deleted then (well I think I did?), however they keep coming back. See the attached image, the two with the red box round them. The two with the green box are also not part of my system and won't delete either. I am removing them the wrong way. Note the only show up the deCONZ gui and not in the API.
[image]

ademobile, 2019-11-01:
Just to add deCONZ is on channel 20 and the two devices showing connection are on another system on channel 11!

ebaauw, 2019-11-01:
ZigBee doesn't do connections.  Each router maintains a neighbour table of known devices.  deCONZ queries these tables to map the network, showing a node for each entry (without checking whether the device is actually reachable).

Where (to what routers) do the lines connect?  These are probably just stale entries in the routers' neighbour tables.  You could try and power cycle the routers (I'm not sure if the neighbour table is supposed to be stored in non-volatile memory), or wait (up to a couple of days?) for the entries to be cleared.

> If so how do you remove them? 

Normally, a device broadcasts a _Leave_ message before leaving the network, so the neighbouring devices remove it from their tables.  Probably not all device firmware does this on a reset (do I recognise some Xiaomi mac addresses?), and maybe some devices don't react well to the _Leave_ message.  Having a multi-vendor ZigBee network is challenging, when not all vendors adhere strictly to the standard.

ademobile, 2019-11-04:
They are both shown connected directly to the coordinator and not through a router. It has been like this for weeks now.

closed 2020-03-10 by stale[bot]