dresden-elektronik/deconz-rest-plugin #2028: can´t pair Hue Lights

wuast94, 2019-11-01:
I Reset The lights with the remote .. the bulbs blink 3 times and switch back on again. but either in phoscon or in deconz shows up anything. 
I power cycled bevor after and bevor and after. but nothing. The Bulb is a few centimeters away.

wuast94, 2019-11-01:
it seems that i can´t pair aqara door sensors and aqara temp sensors too.

wuast94, 2019-11-01:
i updated the conbee firmware but no changes

ebaauw, 2019-11-01:
> i switched the usb place of the stick and since then just the half of my zigbee network showed up

Sounds like radio interference.  Are you using a USB extension cable?  And a USB2 port?  Did you switch off Wifi and bluetooth?  Did you disconnect any USB3 devices?

wuast94, 2019-11-02:
> > i switched the usb place of the stick and since then just the half of my zigbee network showed up
> 
> Sounds like radio interference. Are you using a USB extension cable? And a USB2 port? Did you switch off Wifi and bluetooth? Did you disconnect any USB3 devices?

i switched the stick back where it was before too. so i dont think that this is the problem

wuast94, 2019-11-02:
ok now it works. don´t know why but it works again.

closed 2019-11-02 by wuast94