dresden-elektronik/deconz-rest-plugin #2030: xiaomi cube is not displayed as a switch

salopette, 2019-11-03:
I have vebunden my cube today, but this is not displayed as a switch editor, I have already deleted several times and also restarted Gatway, but no result

[image]
[image]
[image]

Smanar, 2019-11-06:
Do you still have the problem ?
It sem this device is a little shitting to add, but it works on other configuration > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1846

Perhaps if you haven't all sensors created you can see it in phoscon but not in switch editor.

salopette, 2019-11-06:
I have omitted it first. I'm back to the stable, which was with the last ... 70 reboot problems

Smanar, 2019-11-06:
Long time the V70 is out and still not in stable branch, bad sign ^^, I think they have too lot of problem with it.

salopette, 2019-11-06:
Yes, wait, the guys of Zigbee2Mqtt and Zigate have made it! There, however, the Comuniti is bigger!

closed 2020-03-12 by stale[bot]