dresden-elektronik/deconz-rest-plugin #2032: Sensors not creating

bastshoes, 2019-11-04:
Hello!
Just updated to 2.05.70. I'm assuming that release should include LifeControl enviroment sensor support. 
Unfortunately temperature and humidity sensor not appear after rejoining device to network.

bastshoes, 2019-11-05:
Here is the log after joining device to network:
```
16:48:13:839 device announce 0x00158D0000D9BC10 (0xB56F) mac capabilities 0x80
16:48:13:839 set fast probe address to 0x00158D0000D9BC10 (0xB56F)
16:48:13:839 FP indication 0x0000 / 0x0013 (0x00158D0000D9BC10 / 0xB56F)
16:48:13:839                       ...     (0x00158D0000D9BC10 / 0xB56F)
16:48:13:839 device announce 0x00158D0000D9BC10 (0xB56F) mac capabilities 0x80
16:48:14:748 [1] get node descriptor for 0x00158d0000d9bc10
16:48:15:102 FP indication 0x0000 / 0x8002 (0x00158D0000D9BC10 / 0xB56F)
16:48:15:102                       ...     (0x00158D0000D9BC10 / 0xB56F)
16:48:15:102 ZDP indication search sensors 0x00158D0000D9BC10 (0xB56F) cluster 0x8002
16:48:15:102 ZDP indication search sensors 0x00158D0000D9BC10 (0xB56F) clear timeout on cluster 0x8002
16:48:15:108 [2] get active endpoints for 0x00158d0000d9bc10
16:48:16:092 FP indication 0x0000 / 0x8005 (0x00158D0000D9BC10 / 0xB56F)
16:48:16:092                       ...     (0x00158D0000D9BC10 / 0xB56F)
16:48:16:092 ZDP indication search sensors 0x00158D0000D9BC10 (0xB56F) cluster 0x8005
16:48:16:092 ZDP indication search sensors 0x00158D0000D9BC10 (0xB56F) clear timeout on cluster 0x8005
16:48:39:351 APS-DATA.request id: 117, addr: 0x00158D0000D9BC10 profile: 0x0104, cluster: 0x0000, ep: 0x01/0x01 queue: 2 len: 5 (send, fast lane)
16:49:34:387 APS-DATA.request id: 225, addr: 0x00158D0000D9BC10 profile: 0x0104, cluster: 0x0001, ep: 0x01/0x01 queue: 1 len: 5 (send, fast lane)
16:49:43:620 APS-DATA.request id: 61, addr: 0x00158D0000D9BC10 profile: 0x0104, cluster: 0x0201, ep: 0x01/0x01 queue: 2 len: 11 (send, fast lane)
16:49:48:654 APS-DATA.request id: 114, addr: 0x00158D0000D9BC10 profile: 0x0104, cluster: 0x0402, ep: 0x01/0x01 queue: 3 len: 11 (send, fast lane)
16:49:49:663 APS-DATA.request id: 118, addr: 0x00158D0000D9BC10 profile: 0x0104, cluster: 0x0402, ep: 0x01/0x01 queue: 1 len: 11 (send, fast lane)
16:50:29:815 APS-DATA.request id: 150, addr: 0x00158D0000D9BC10 profile: 0x0104, cluster: 0x0402, ep: 0x01/0x01 queue: 3 len: 11 (send, fast lane)
16:50:30:830 APS-DATA.request id: 158, addr: 0x00158D0000D9BC10 profile: 0x0104, cluster: 0x0402, ep: 0x01/0x01 queue: 1 len: 11 (send, fast lane)
16:50:34:848 APS-DATA.request id: 197, addr: 0x00158D0000D9BC10 profile: 0x0104, cluster: 0x0000, ep: 0x01/0x01 queue: 1 len: 5 (send, fast lane)
```

bastshoes, 2019-11-12:
Hello @manup!

Why do sensors not creating? Is it because device not reporting info for 8002 and 8005 cluster in time?
Is there anything we could do on deconz side?

closed 2020-03-18 by stale[bot]

SwoopX, 2020-04-29:
So, you've physically reset the device, but the sensors are not created during join? You used Phoscon for seansor search and it did not return the green success message?

If you can compile the plugin for testing, I can make you a version available that has quite some additional debugging for that.

bastshoes, 2020-04-30:
Yes, I've rest device. I see via VNC that it joined network.  Green success message not appears.

Yes can try to compile plugin.

LouiS22, 2020-10-25:
Hello, any update on this? If needed, I can test anything, I have this sensor.

Mimiix, 2020-10-26:
@LouiS22 I think it's best to open a new device request.