dresden-elektronik/deconz-rest-plugin #2033: 2.05.70 does not work with -platform minimal

Flurkmark, 2019-11-04:
It does not work headless. I build from repo yesterday and did not get it to work. Works with X forwarding, which I can only temporary use. Thought it was something wrong with the my build environment. Downloaded .deb today, same thing. Locks and needs to be killed, when looking with strace just sitting waiting on a futex. 

Debugging info gave little, locks output at different places. 

Copying old libde_rest_plugin.so gets it working again.

manup, 2019-11-04:
Just tried it, on my Raspberry Pi 4 with Raspbian Buster headless works.

Can you please check the output of:

    deCONZ -platform minimal --http-port=8080 --dbg-error=1 --dbg-info=1

lbouriez, 2019-11-04:
Hello, there is no release informations regarding this version, will it come soon :) ?

Flurkmark, 2019-11-04:
Interesting.
```
pi@raspberrypi:~ $ deCONZ -platform minimal --http-port=8080 --dbg-error=1 --dbg-info=1 
```
I cut and pasted your line. 
Actually it started the gui version, since I had X forwarding on. From another terminal same client:
```
qt.qpa.screen: QXcbConnection: Could not connect to display 
Could not connect to any X display.
```
This is exceptionally wierd because if I open a terminal from another raspberry running KDE, I don't even get that error message and deconz needs to be killed. From my workstation I get clean shutdown after error message. Very odd. Must be som env at play.
fish://root@192.168.1.3/home/pi/out.txt

manup, 2019-11-04:
>Hello, there is no release informations regarding this version, will it come soon :) ?

Yes :)

manup, 2019-11-04:
>I cut and pasted your line.
>Actually it started the gui version, since I had X forwarding on. From another terminal same client:

That's really strange, the -platform minimal shouldn't start a gui.

Is this vanilla version or self compiled plugin?

manup, 2019-11-04:
Is the service disabled?

    systemctl status deconz
    systemctl status deconz-gui

Flurkmark, 2019-11-04:
This is the vanilla version from your .deb
Service stopped and verified. 

I used to be a developer some decade ago, so I’m a bit rusty. Can some qt version screw it up?

How do I check out the .69 version with git?
Would be interesting to see if the plugin still is okay if I build it myself. 

By “replacing the plugin” I mean that I copy the .so file from the .69 deb to plugins/ , starts right up.

Flurkmark, 2019-11-05:
Did some more testing. If I compile .69, still works. 

If I don’t use -platform minimal, it gives up straight away as expected. 
If I do it get’s to ‘ZCL attribute report’ and shuts down rest api, as in earlier log file. So, not build environment then...

Flurkmark, 2019-11-06:
@manup The bug is introduced in 59b7a04

manup, 2019-11-06:
Interesting, thanks for digging into this, from a quick look I can't spot why the commit is causing the problem. A colleague had a problem with touchlink today which didn't occur in .69, might be related.

Flurkmark, 2019-11-07:
@manup It's difficult to say for me, since I don't know the code and don't have the source for the launcher, can't run it in a debugger. 

Something happens when it tries to reconnect. It starts looking for ttyS0 and so on. When finally connecting to ttyUSB0 it triggers. I am wildly guessing that maybe the Qapp restarts and the option -platform which is not a standard longopts gets tossed. 

Is there a option for device? --device= does not work.. 

Anyway, if I rename ttyUSB0 to ttyS0 so it connects straight away, the bug is not triggered. 
Now running .70 without X for a hour, with renamed /dev without issues.

manup, 2019-11-07:
>I am wildly guessing that maybe the Qapp restarts and the option -platform which is not a standard longopts gets tossed.

Yes I think that's the case too, will do some tests tomorrow and hopefully find a fix.

manup, 2019-11-17:
2.05.71 is building now and should be online by tomorrow, hope this fixes the issue.

closed 2020-03-23 by stale[bot]