dresden-elektronik/deconz-rest-plugin #2034: Timeout when trying to access web UI

VivantSenior, 2019-11-05:
Hi,

I have an issue with accessing the ConBee 2 WebUI.

My configuration is:
- Raspberry Pi 3B+ and Raspbian
- Home Assistant 0.101.2 (hass.io)
- Hass.io supervisor 192
- I have GBAN stick plugged in,
- ConBee 2 stick
- deCONZ 3.6 add-on

Config of deCONZ:
```
{
  "device": "/dev/ttyACM0",
  "vnc_password": deconz_vnc_pass_here,
  "dbg_info": 1
}
```

Ports:
```
5900/tcp **disabled** deCONZ graphical desktop via VNC
80/tcp **40850** deCONZ API backend
8080/tcp **40860** deCONZ WebSocket Server
```

After:
- forwarding port 40850 on the router,
- starting the addon,
- trying to access the “GUI interface” (http://xxx.duckdns.org:40850/pwa/index.html)

I’m receiving a timeout error.

How can I access the web GUI? (if I enable the VNC port I’m able to access VNC server, but no luck with web GUI)

Thanks,
Viv

VivantSenior, 2019-11-05:
Partially resolved. I have HTTPS Everywhere extension which redirects from http to https connections. deCONZ is trying to connect me via http (don’t know why - I have Let’s Encrypt certificates) but got redirected to https and therefore it times out…

The connection should be https, instead of http.

manup, 2019-11-05:
deCONZ currently doesn't support HTTPS this is only possible by using a reverse proxy like nginx in between.

VivantSenior, 2019-11-07:
@manup, ok so I would like to propose it as a feature request because it's quite a serious security vulnerability.

closed 2019-12-02 by VivantSenior