dresden-elektronik/deconz-rest-plugin #2035: Lost connection with all devices -> no control of devices

msvds, 2019-11-05:
Hi, I am using DeConz on Hassio now for 1 week. Before I was using zigbee2mqtt and switched to Deconz because of its better range. In two days during the weekend, I reset and paired most of my devices and indeed the range is better than with zigbee2mqtt using the CC2531 stick. Pretty nice result after several hours of pairing everything.

Unfortunately, in the night from sunday to monday suddenly all the connected devices from Deconz were gone in Hassio. Later I checked in Phoscon and also here the devices were still there, but not able to control them anymore.

I am using these type of devices:

- IKEA Tradfri GU10 bulbs (10)
- IKEA Tradfri E27 bulbs (2)
- IKEA Tradfri control outlets (2)
- Osram Classic A60 W clear - LIGHTIFY (2)
- Gledopto GL-B-007Z (1)
- Xiaomi switches (wireless, wired without neutral, smart switch)
- Xiaomi motion sensors (10)
- Xiaomi door/window sensors (11)
- Xiaomi humidity/temperature sensors (7)

I am using DeConz 3.6 on Hassio 0.101.2 which runs on a virtual machine on a QNAP NAS. I use a Conbee 2 stick with firmware 2.05.69 (264A0700).

I have tried to load the zigbee netwerk configuration in Phoscon (using alt-Advanced tab), but that didn't help anything.

I would like to know if there is another option then resetting the gateway and repairing all the devices. Because this will again require several hours of work. And the last time I did that, it worked for less than 1 day. Please help me to find these reason of this.

Let me know if you need more information.

Here is the log file:

starting version 237

[10:27:39] INFO: Starting VNC server...
[10:27:43] INFO: Starting the deCONZ gateway...
[10:27:43] INFO: Running the deCONZ OTA updater...
[10:27:43] INFO: Running the IKEA OTA updater...
[10:27:43] INFO: Running Hass.io discovery task...
{"uuid":"4e1bcf2a947845029af1aa896b126fe9"}
[10:27:43] INFO: Success send discovery information to Home Assistant
[10:27:43] INFO: deCONZ is set up and running!
libpng warning: iCCP: known incorrect sRGB profile
10:27:43:790 HTTP Server listen on address 0.0.0.0, port: 40850, root: /usr/share/deCONZ/webapp/
10:27:43:973 CTRL. 3.22.010:27:44:288 COM: /dev/ttyACM0 / serialno: DE2119370
10:27:44:289 COM: --dev: /dev/ttyACM0 (ConBee II)
10:27:44:289 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt
10:27:44:457 parent process bash
10:27:44:459 gw run mode: docker/hassio
10:27:44:461 GW sd-card image version file does not exist: /data/.local/share/dresden-elektronik/deCONZ/gw-version
10:27:44:463 DB sqlite version 3.22.0
10:27:44:468 DB PRAGMA page_count: 45
10:27:44:468 DB PRAGMA page_size: 4096
10:27:44:468 DB PRAGMA freelist_count: 2
10:27:44:469 DB file size 184320 bytes, free pages 2
10:27:44:469 DB PRAGMA user_version: 6
10:27:44:469 DB cleanup
10:27:44:470 DB create temporary views
10:27:44:505 don't close database yet, keep open for 900 seconds
10:27:44:507 started websocket server at port 40860
10:27:44:517 found node plugin: libde_rest_plugin.so - REST API Plugin
10:27:44:534 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
10:27:46:046 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
10:27:46:126 COM: /dev/ttyACM0 / serialno: DE2119370
10:27:46:127 COM: --dev: /dev/ttyACM0 (ConBee II)
10:27:46:357 COM: /dev/ttyACM0 / serialno: DE2119370
10:27:46:358 COM: --dev: /dev/ttyACM0 (ConBee II)
10:27:46:454 COM: /dev/ttyACM0 / serialno: DE2119370
10:27:46:454 COM: --dev: /dev/ttyACM0 (ConBee II)
10:27:47:260 wait reconnect 15 seconds
10:27:47:429 COM: /dev/ttyACM0 / serialno: DE2119370
10:27:47:433 COM: --dev: /dev/ttyACM0 (ConBee II)
10:27:48:261 wait reconnect 14 seconds
10:27:49:260 wait reconnect 13 seconds
10:27:50:186 Announced to internet
10:27:50:259 wait reconnect 12 seconds
[... 244 lines cut ...]
10:28:56:825 APS-DATA.indication request id: 225 -> finished
10:28:56:826 APS-DATA.request id: 225 erase from queue
10:28:56:826 ZDP status = 0x00 -> SUCCESS
10:28:56:826 ZDP Mgmt_Lqi_rsp zdpSeq: 9 from 0x00212EFFFF04D7C7 total: 1, startIndex: 0, listCount: 1
10:28:56:827     * neighbor: 0x00158D00022B85AC (0x4D21), LQI: 255, relation: 0x01 rxOnWHenIdle: 1
10:29:06:007 aps request id: 14 prf: 0x0000 cl: 0x0000 timeout (confirmed: 1) to 0x90FD9FFFFE105696 (0xFFFD)
10:29:06:088 aps request id: 14 finished, erase from queue
10:29:11:688 APS-DATA.request id: 30, addrmode: 0x02, addr: 0xfffd, profile: 0x0000, cluster: 0x0000, ep: 0x00 -> 0x00 queue: 1 len: 11 tx.options 0x00
10:29:11:688 send NWK_addr_req to 0x086BD7FFFE55C831, last seen 0 s, last seen by neighbors 69 s
10:29:12:350 APS-DATA.confirm id: 30, status: 0x00 SUCCESS
10:29:21:767 Mgmt_Lqi_req zdpSeq: 11 to 0x00212EFFFF04D7C7 start index 0
10:29:21:768 APS-DATA.request id: 72, addrmode: 0x03, addr: 0x00212effff04d7c7, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x00
10:29:21:775 APS-DATA.confirm id: 72, status: 0x00 SUCCESS
10:29:21:776 APS-DATA.confirm request id: 72 -> confirmed, timeout 1572946161
10:29:21:783 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 241, rssi: 19
10:29:21:784 APS-DATA.indication request id: 72 -> finished
10:29:21:784 APS-DATA.request id: 72 erase from queue
10:29:21:784 ZDP status = 0x00 -> SUCCESS
10:29:21:784 ZDP Mgmt_Lqi_rsp zdpSeq: 11 from 0x00212EFFFF04D7C7 total: 1, startIndex: 0, listCount: 1
10:29:21:785     * neighbor: 0x00158D00022B85AC (0x4D21), LQI: 255, relation: 0x01 rxOnWHenIdle: 1
10:29:22:728 ZDP node descriptor request to 0x00158D00022B85AC
10:29:22:729 APS-DATA.request id: 77, addrmode: 0x02, addr: 0x4d21, profile: 0x0000, cluster: 0x0002, ep: 0x00 -> 0x00 queue: 2 len: 3 tx.options 0x04
10:29:22:729 ZDP send request id: 0x03 to 0x00158d00022b85ac
10:29:23:050 aps request id: 84 prf: 0x0000 cl: 0x0000 timeout (confirmed: 1) to 0x00124B001BD841F6 (0xFFFD)
10:29:23:128 aps request id: 84 finished, erase from queue

ottelo9, 2019-11-06:
I've the same problems at the moment with all my components. I'm using Conbee II and Home Assistant on Rpi3+ for months now. 1-2 months ago I lost the connection to all my components. I restart the Pi, Ha, Deconz, reinsert the ConBee Stick, reimported my conbee backup and so on. But nothing helps. ONLY a hard power off of all my light bulbs works! And no the same happens again, since 9 AM all components went offline. I check the connection status via VNC (see pic). The stick lost connection to all devices (no lines any more!). So I tried to leave network and rejoin, but rejoin doesnt work!!

[https://www.bilder-upload.eu/bild-d872a1-1573054698.png.html](url)

luci84tm, 2019-11-08:
Hi there,
I face also a problem after switching from pi3 to pi4 docker deconz, many of the devices do not join the network:
image
[image]

I'm running the latest version:
image
[image]

What can I do without resetting everything ?

SwoopX, 2019-11-09:
Same here. I updated from .65 to .69 one week ago and 5 days after update, have my network was dead. Interestingly, only my innr plugs went MIA (and of course the associated sensors). Really curious was the ocasionally, I have 2 plugs next to each other, one staying operational. I also sniffed the zigbee traffic for about 2h and nothing from the lost devices (nothing sent from the devices or any responses). Have resetted the devices after all, not exactly the fun part.
First time this has ever happened with deconz since I started 2 years ago.

luci84tm, 2019-11-11:
@SwoopX resting all devices fix the problem, or you still have some which are not connected?
How does your network look like after reset?

I would like to get some confirmation, before I reset everything and update all my automations + devices in HA.

SwoopX, 2019-11-11:
@luci84tm well, things went back to normal so far and everything is back online. Lost my consumption counters but my daily consumption is calculated in the middleware, so nothing serious. Simple power cycle hasn't helped.

Good thing is that, if you don't delete the end devices in Phoscon or deconz GUI, they retain their IDs after reset. The sensors also came back without reset or re-pair. Afaik, some devices support some kind of re-pairing without reset. You might want to try that first if it is supported.

luci84tm, 2019-11-14:
Resetting is not really working for me..
Device ca no longer join the network.
My PI +Raspbee are near my wifi router, for zigbee I use channel 11.
Can it be that channel overlapping with wifi and that's why I have zigbee network problems?

ottelo9, 2019-11-14:
> Resetting is not really working for me..
> Device ca no longer join the network.
> My PI +Raspbee are near my wifi router, for zigbee I use channel 11.
> Can it be that channel overlapping with wifi and that's why I have zigbee network problems?

Please have a look at this page: https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html

closed 2020-03-20 by stale[bot]