dresden-elektronik/deconz-rest-plugin #2036: Zigbee network key & channel

VivantSenior, 2019-11-05:
Hi,

is it possible to add [Zigbee Network Key](https://www.zigbee2mqtt.io/how_tos/how_to_secure_network.html) and change [Zigbee channel](https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html) to 26?

Best,
Viv

Smanar, 2019-11-06:
Have you take a look in deconz, in the GUI ? In "network setting"

VivantSenior, 2019-11-07:
@Smanar, thank you. BTW if I were you I would gray out the Network key value.

VivantSenior, 2019-11-08:
I have set the channel to 26 in VNC but after a few hours and resets of deCONZ add-on I can see in the logs:
```
Current channel 25
```

Smanar, 2019-11-08:
Lol, I m not paranoid, this screenshoot is from google, but you realy think someone can note this key, come in my country, in my city, close to my house, sniff my traffic to see my living room temperature or turn on the light when I m sleeping ? It's just zigbee.

For the channel 26, I can't make tries on my side, this channel is special, not autorised on somes countries by legislation, so perhaps It have something special. Try with another, if it change, there is something on channel 26, else your procedure to change channel  is bad.

Digital-Thor, 2019-11-19:
The chart at https://github.com/dresden-elektronik/deconz-rest-plugin/issues/825#issuecomment-427139292 may help you pick your Zigbee channel.  When I went to change mine (in Phoscon), it warned me of proceeding, but it seems hard to understand why I would want to be at the current setting of channel 11 in North America!  Aren't Zigbee coordinators supposed to pick the quietest channels?  Or is that just a one-time static setting that doesn't adjust if conditions change?

asiantuntija-asiantuntija, 2019-12-20:
I've been using channel 26 to get as far away as possible from Wifi channels since early 2019. Haven't seen anything odd so far when using it nor while adding new devices. I'm using battery powered Aqara wall switches, Aqara water leak sensors, Aqara door sensors, Gledopto LED controllers, Philips Hue lamps and Ikea repeaters all on channel 26 without issues. I used to have some Osram power outlet switches, but regardless what channel I used those caused issues with Aqara wall switches losing first button press after coming back from power save mode. So Osrams worked on channel 26 too, but had other compatibility issues.
 
Trying to change channel over VNC using X11 app didn't work for me. To enable channel 26 I patched this check away to allow channel 26 and then used REST API with curl to do actual channel switch. Change has persisted over version upgrades so no need to re-do patch each time software is updated.
 
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/rest_configuration.cpp#L1898

Dec 20 23:51:13 zigbee deCONZ[12241]: 23:51:13:233 Device firmware version 0x26320500
Dec 20 23:51:14 zigbee deCONZ[12241]: 23:51:14:369 Current channel 26

On older ticket someone speculated that channel 26 might be restricted due regulatory or compatibility reasons. Whatever reason is, currently only channels 0, 11, 15, 20 and 25 are allowed without resorting to hacks.

Cyrille63, 2020-03-09:
Hi I need to use temporarily this channel 26 to change default channel of my livolo switches

Livolo switches are by default on channel 26
I must pair the switch on channel 26 and then I change my channel tou 15
The switch will pair now on channel 15

I must patch the code if I want to use channel 26 ? No other solution ?

Smanar, 2020-03-09:
Have you the GUI.

I m not sure but from my memory only the API can prevent some channel, if you change it with the GUI, the API don't chnage it again.

But idk if all users have it.

[image]

Cyrille63, 2020-03-09:
Yes I have the GUI
I will test and give you the answer 
Thks

Cyrille63, 2020-03-14:
In Network Settings, I uncheck channel 15 and check channel 26 but when I save, it checks channel 15 and uncheck channel 26.

So I can not make my network on channel 26.

I need help, Cyrille

Cyrille63, 2020-03-21:
I succeded to change channel 26 and after channel 15. I can see Livolo switch in deCONZ but Home Assistant deCONZ plugin fails to find the livolo switch
Help please, thanks in advanced

Smanar, 2020-03-21:
You can see it in deconz, phoscon, or in REST API ?
If you can see it in phoscon or in REST api, it must come from HA.

Cyrille63, 2020-03-21:
I see it in deCONZ
[image]

Smanar, 2020-03-21:
OK it don't seem to be in the API.

But it's not a wired device ?

BTW, the pairing is not finished yet, you miss the second circle.

Cyrille63, 2020-03-22:
No it's not a wired device

Cyrille63, 2020-03-22:
On channel 26 I have the second circle but can not paired in phoscon ou webapp
[image]

Cyrille63, 2020-03-22:
Webapp test
[image]

Smanar, 2020-03-22:
Ok so it's better.
But you have a wireless version ?

To try to include it I need theses screenshoot > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

Cyrille63, 2020-03-22:
Sorry the livolo switch is a wifi zigbee switch. Here is the reference VL-C701Z

https://www.livoloeurope.eu/livolo-zigbee-1gang-1way-touch-light-switch-vl-c701z

Smanar, 2020-03-22:
Hur, endpoint and cluster are bad, realy bad device, will be hard to include.

Can you send the screenshoot ?

Cyrille63, 2020-03-22:
No cluster info
[image]

Smanar, 2020-03-22:
Ok so it's that this device have bad MAC capabilities.
For cluster info.
- enable cluster info panel
- select Basic cluster 0000
- you will have the table in the bottom.

Now, if I make some code modification, can you compile it to test ? (you need an Unix manchine) ?

Cyrille63, 2020-03-22:
Here is cluster info

I run ConBee II on Windows 10 machine, can I compile on a virtual linux machine but for Windows ?

[image]

Smanar, 2020-03-22:
Nope :(

Np I will try to find someone.

Smanar, 2020-03-22:
BTW, you have re-include your device since the last time ?

- the "led on node" is green.
- the NWK have changed 0x4417 > 0xCE07

Cyrille63, 2020-03-22:
Yes, I do lot of tests and I reset the Livolo switch

Smanar, 2020-03-22:
Ok, because I know a little this device from zigate and zigbee2mqtt and it s realy a crappy device.
I want to be sure it s stable in deconz before trying to include it.

MOE-JOE2, 2020-03-29:
I bought a Livolo Switch + Deconz adapter. Unfortunately I'm not a developer, if there's a guideline howto compile that project on linux I can do that + I am able to provide you some more debug logs if needed. Just let me know.

Smanar, 2020-03-30:
Sure, you have working mode on the homepage of the github > Install deCONZ development package (optional, Linux only)

I will try some  modification and give you a link for the news files to compile.

I just need to be sure the device is completely paired with deconz.
- Stable Nwk ID
- No green led on node

And for the moment it' the only thing you can test, because you haven't cluster to test the device working mode.

Cyrille63, 2020-04-02:
I have install a linux machine - Lubuntu 18.04.04 - deCONZ is working - I have installed hte development package, compiliation is OK

I can do test

Smanar, 2020-04-02:
Just tell me if you have successfully include the device in deconz (even it don't appear in phoscon)
I need the device have a stable Nwk ID, and no green led on  ode title.

If all is ok on your side, I can make some code modification today for test.

Cyrille63, 2020-04-02:
Yes, the device is include in deconz
Yes, the network ID is stable
No green LED

Sorry for the double comment

Smanar, 2020-04-02:
@SwoopX, by chance, you haven't see this kind of situation on deconz ?

I m searching a way to by-pass the mac capabilities of the device or adding myself missing cluster.
I need to add the cluster 0006 but in input, to try if the device can react to command trought GUI.

I m sure I have already see that on the github, but I don't find the device.

SwoopX, 2020-04-02:
Maccapabilities is 0? Yes. Search for develco in rest_sensors.cpp, you can bypass it there.

Cyrille63, 2020-04-02:
Yes it's a wired switch. No battery. I thought the question was about the network connection

Cyrille63, 2020-04-03:
I try to pair like SwoopX said by Phoscon but it does not work

Cluster info 0006 does not want to be blue

0000
[image]

0003
[image]

0006
[image]

Smanar, 2020-04-03:
You have missed ^^.
All fields are empty.

Nevermind, I will see if I can force the state/on in the sensor and make try as it.

Smanar, 2020-04-03:
Ok so Fail, not possible use state/on for sensor.
We need to add in light section.

Now How to force a sensor to be in light device ....

Cyrille63, 2020-04-03:
> You have missed ^^, you have pressed the "read" button ?

There were no read button

Smanar, 2020-04-03:
Here
[image]

Ok, so first try, nothing special, I have just add the device in whitelist
To compile it, make like on the homepage but in step 1, use

` git clone https://github.com/Smanar/deconz-rest-plugin.git`

Smanar, 2020-04-03:
You have do that with the modified version from my fork ?

For the white list I m using
Mac adress 0x00124bXXXXXXX
Model id : TI0001
Manufacture code 0x0000

And I don't think theses values have changed ?

Cyrille63, 2020-04-03:
Yes with your fork
As you can see, there is now brand and model ID in the cluster info 0000. I suppose it's because your modifications.

Smanar, 2020-04-04:
Nope you already had them https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2036#issuecomment-602201388

Can you make a try, with keeping the device paired, set again Phoscon in permit join, on cluster 0000 Basic, read again the attributes, and recheck for new deive in API.

If it still not working, can you make the inclusion like usually but with deconz log enabled ? > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/55

Cyrille63, 2020-04-04:
Sorry for my poor english, I wanted to say it was the first time I saw the model/brand of my switch in deCONZ (to prove I use you development)
[image]

First test -> nothing better, no new device

Second test -> [Log-2.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4431552/Log-2.pdf)

Smanar, 2020-04-04:
Ha ? Oups sorry, so it's a good news It mean the API reconize the device.
But I have put a log to explain wich one command line to use
`deCONZ --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1`

But take care the log can be huge, so better to send output to console, and make a copy/paste, I just need the log that start when you have include the device up to 1/2 mn after.

To don't have to include the device each time, there is this magic trick (phoson in permit join + read basic attribute), but on your log you need to see something like that

>07:55:52:004 device announce 0x0004740000A132F6 (0x6DC7) mac capabilities 0x8E

Cyrille63, 2020-04-07:
Here is a new log -> [Log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4444724/Log.txt)

I ran : deCONZ --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1 > Log.txt

Nothing in sensor, nothing in switch, nothing in light

Smanar, 2020-04-07:
Hello,
Have you set Phoscon in permit join (and not deconz)?
I don't see the permit join in the logs
>18:38:20:516 send permit join, duration: 59

It seem the inclusion procedure haven't started at all ? I see
>ZDP device announce

From deconz, but nothing after from the API.

Cyrille63, 2020-04-07:
Hello,
Oups, here logs with Phoscon API -> [Log-2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4445216/Log-2.txt)

Cyrille63, 2020-04-07:
Nothing
[image]
[image]
[image]
[image]

Smanar, 2020-04-08:
Nope ^^ in Phosocn don't mean in the API.
Phoscon is using the API too, but It choose what it want to be displayed.

To check the api the easy way is using your web browser with something like
http://IP:PORT/api/KEY/sensors
http://IP:PORT/api/KEY/lights

BTW are you using an application with deconz to get an api key ? Else you have all informations here https://dresden-elektronik.github.io/deconz-rest-doc/ but there is easier way.

Note; wired switch as set as "light".

Cyrille63, 2020-04-09:
I can see it in API in sensors but not in ligths 

Sensors
[image]

Ligths
[image]

Smanar, 2020-04-10:
Ok, so I have some problem ATM, I m on 2 devices in same time.

So can you just edit your de_web_plugin.cpp file with this one https://github.com/Smanar/deconz-rest-plugin/blob/livolo/de_web_plugin.cpp
It's on a livolo branch, so not exctly the same command than previous ?

If you have problem with line 1655
Juste replace it by

>if ((node->nodeDescriptor().manufacturerCode() == VENDOR_NONE) && (true) )

Cyrille63, 2020-04-10:
Yes !
Now it's a light device and it appears in Phoscon
[image]
[image]

Smanar, 2020-04-10:
And it have the good field "State / on / False.
Now the more sensible part. Can you change the switch state using Phoscon or the API ?

With the API and curl it will be something like 
curl -H 'Content-Type: application/json' -X PUT -d '{"on": true}' http://PORT:IP/api/KEY/lights/3/state

honestly IDK what it will happen, error message perhaps, because this device is special.

Cyrille63, 2020-04-10:
Yes I can change state in Phoscon the icon change color yelow/gray

In the API
[image]
[image]

Smanar, 2020-04-11:
Ok, so It's something normal, I have more information from Pipiche
This device use Setlevel command for on/off and using special value., So I m making a special part just for him.

I have made a new version > https://github.com/Smanar/deconz-rest-plugin
You can use git pull to update your files.
The device is already included, so you juste need to close deconz, update the file, reload deconz.
If you want to re-include the device you need to edit the line 1655 in de_web_plugin.cpp, just replace false by true, but if your device is already included, it's not a problem.

The code is realy experimental, but if I m right it will work with Phoscon.
Perhaps only off or only on will work, so you can trigger the device manualy.

Cyrille63, 2020-04-11:
It works in Phoscon for a small time
- I can switch on the ligth by click on the ligth **1**
- I can switch off the ligth by click on the group **2**
- I can do that several times but after nothing

[image]

I don't know it's normal but the switch is seen as light and sensor at the same time
Before my test I delete all the devices and I do a re-inclusion
lights
[image]
sensors
[image]

Smanar, 2020-04-11:
Yep, this device is declared as battery sensor, so it have a sensor entry, but to send it order we need to use a light entry, so you have 2 entries.

We can prevent the sensor creation later, but my first objective is making the device working.

I can switch on the ligth by click on the ligth 1 > It mean you can't set it off using light 1 ?
I can do that several times but after nothing > It mean after some tries, it stop working ?

BTW, I haven't set group woring for the device yet, but I think It will not work.
When you use group, the conbee send a command to all devices, the same command, and this device need special command, different than for other device.
And I need to check how you can set it off, using the group feature ....

Cyrille63, 2020-04-11:
OK thanks for the explanation of the sensor in Phoscon

> I can switch on the ligth by click on the ligth 1 > It mean you can't set it off using light 1 ? 
Yes I can not set off using light 1

> I can do that several times but after nothing > It mean after some tries, it stop working ?
Yes it stops working

Smanar, 2020-04-12:
Ok So I have checked the code agin, and I don't see the problem for 'off'
Something you can try, in file rest_light.cpp in function setLivoloDeviceState() line 1571

Re-enable this part at the function bottom (remove the //)
```
    if (taskRef.lightNode)
    {
        updateLightEtag(taskRef.lightNode);
        rsp.etag = taskRef.lightNode->etag;
    }
```

and for off part 
```
            if (isOn)
            {
                ok = addTaskSetBrightness(task, 0x6c, false);
            }
            else
            {
                ok = addTaskSetBrightness(task, 0x01, false);
                //ok = addTaskSetOnOff(task,ONOFF_COMMAND_OFF,0)
                //ok = addTaskSetOnOff(task,ONOFF_COMMAND_TOGGLE,0)
            }
```

You can try the 2 lines it send the sam command then for group, or the 3th, it's a toogle command, disable the 2 others when you enable one.

What are you doing when it stop working, it work again after some time ?

[58 more replies]

closed 2020-06-06 by Mimiix