dresden-elektronik/deconz-rest-plugin #2037: Magic Cube (Mi or Aqara) useless the way it is presented in deconz event in Home Assistant

KennethLavrsen, 2019-11-05:
I am not sure where the issue is.
But when you add a Magic Cube to Deconz it shows as a device with State and a rotation.
But in Home Assistant Deconz Event there is no difference between the states and the rotation. You cannot tell if the event is move of tilt or share or it is a rotation event. It is just a single number.

So I cannot create any kind of automation in Home Assistant because the event name is the same and the event is just a number. We need some way to see the difference between the 4 digit event numbers and a rotation Either two different device names or some other means. As it works now I cannot use a Cube for anything in Home Assistant.

KennethLavrsen, 2019-11-05:
I raised same issue on Home assistant github as it is not clear to be where the problem is and where it should be resolved

https://github.com/home-assistant/home-assistant/issues/28572

KennethLavrsen, 2019-11-05:
I have some additional info.

I have noticed something new.

I deleted my Magic Cube and cleaned manually the core.device_registry and core.entity_registry.

And then restarted HA and re-paired the cube in Phoscon

It then gets recorded as switch_81 and switch_82. switch_81 is the name for the battery sensor also.

Then I restart Home Assistant and the switch_82 autorenames (I did nothing in Phoscon) to Mi Magic Cube (the default name that Phoscon assignes). The battery sensor stays as switch_81 in HA.

Now turning the cube is switch_82. And rotation are events from mi_magic_cube.

Then I rename the device in Phoscon and all events become the new name and you cannot separate the events again. The problem is that the Cube gets registered as two different switches by default but the minute you rename the cube to something else in Phoscon else merge to one.
Note that the problem with same name happens after a new restart of HA. HA sync the device name from Deconz only when you restart HA
If the event names had a suffix auto appended it would work and you could seperate the two types of events

KennethLavrsen, 2019-11-05:
Update of update. Yes. If you accidently rename the device in Phoscon then delete it in Phoscon and add it again. Problem is that now the device number increase and you can rewrite your automations in Home Assistant again for non-rotate events. Mine are now called switch_84. It is a work around but a terrible work around. I hope either the Deconz or HA side of the fense come up with a solution

ebaauw, 2019-11-05:
I don't know HA, but I think the issue is how HA identifies the devices.  You don't want to use the name, since there's no guarantee that that's unique, and it might be changed by the user.  Also, you don't want to use the resource id, since that can change when the device is removed and re-paired.

In homebridge-hue, I use the `uniqueid` attribute, which is both unique and unchangeable for resources corresponding to ZigBee devices.  It's composed of the ZigBee mac address, endpoint, and (for sensors) cluster.  It also allows me to recognise the device still, when moved from e.g. a Hue bridge to deCONZ.

Note however, that user-defined resources (CLIP sensors, groups, ...) don't have a `uniqueid` that's guaranteed to be unique.  For these I do use the resource id.

KennethLavrsen, 2019-11-06:
To those needing a work around while this item remains open.

See this forum post. It makes the cube possible to use with current code

https://community.home-assistant.io/t/mi-magic-cube/136701/24

Not elegant but it works

KennethLavrsen, 2019-11-06:
@ebaauw 
This is the API reply I get from Deconz related to the Cube

```
      "83":{ 
         "config":{ 
            "battery":100,
            "on":true,
            "reachable":true,
            "temperature":2200
         },
         "ep":3,
         "etag":"920fef0342ea96a8e8bb9495b86604af",
         "manufacturername":"LUMI",
         "mode":1,
         "modelid":"lumi.sensor_cube.aqgl01",
         "name":"Mi Magic Cube",
         "state":{ 
            "buttonevent":7677,
            "gesture":7,
            "lastupdated":"2019-11-06T09:48:08"
         },
         "swversion":"20160704",
         "type":"ZHASwitch",
         "uniqueid":"00:15:8d:00:02:b9:33:a9-03-000c"
      },
      "84":{ 
         "config":{ 
            "battery":100,
            "on":true,
            "reachable":true,
            "temperature":2200
         },
         "ep":2,
         "etag":"799f7fe8ead04dd0012a3da01620ade8",
         "manufacturername":"LUMI",
         "mode":1,
         "modelid":"lumi.sensor_cube.aqgl01",
         "name":"Switch 84",
         "state":{ 
            "buttonevent":6002,
            "gesture":3,
            "lastupdated":"2019-11-06T09:47:50"
         },
         "swversion":"20160704",
         "type":"ZHASwitch",
         "uniqueid":"00:15:8d:00:02:b9:33:a9-02-0012"
      },
```
This is how things look like after pairing the Cube but before renaming.

The problem is that when you rename the Cube in Phoscon the same name is stored in both devices

And in Home Assistant they have decided to only report the "buttonevent" and "lastupdated"

And all other IDs in the JSON in the HA events are identical between the two devices

There are several things that can help. My workaround in previous post is a renaming of the two overlapping devices via Rest which is working but a bit of a hack. A more user friendly way to have two different names reported is one way. But having two devices in Phoscon will make Phoscon more complicated for normal users. You could handle the Cube as a special case and append a suffix

A better way may be to enhance the HA integration so it includes the "gesture" so you can condition automations on that.

A third way would be if the events for rotation were reported in degrees that are always max 3 digits. You cannot use the decimal number for anything. I would be happy to see a number of degrees in the interval -360 to 360 and then it would always be different from the 4 digit button events. You can even pad it with leading zeros if the 4 digits are important. That would be a breaking change though!

All the HA Deconz user needs is some way to see the difference between the rotation and the other modes.

ebaauw, 2019-11-06:
> This is the API reply I get from Deconz related to the Cube

Yes, that's by design: one device with two resources, `/sensors/84` for the _Multistate Input_ cluster on endpoint 0x02 (notice the `uniqueid`) and `/sensors/83` for the _Analog Input_ cluster on endpoint 0x03.  The _Mutistate Input_ cluster reports the "digital" events (flip, push, drop, etc); the _Analog Input_ cluster the rotation events.

> The problem is that when you rename the Cube in Phoscon the same name is stored in both devices

From the deCONZ REST API perspective, that's not a problem. `name` isn't supposed to be unique.

> A third way would be if the events for rotation were reported in degrees that are always max 3 digits. 

I like this idea.  Currently we simple expose the raw value reported by the cube, basically because I don't know how to scale this to arc degrees.

> That would be a breaking change though!

If we're making breaking changes, we probably want to expose this cluster as a (new) ZHARelativeRotary `/sensors` resource, using `state.rotaryevent`, cf. the ZLLRelativeRotary introduced in v1.30 of the Hue API (see section 4.4 on https://developers.meethue.com/develop/hue-api/supported-devices/).  I don't know what device this is, though; the sample JSON is fabricated (look at the mac address in the `uniqueid`).

Thomas-Vos, 2020-01-20:
> I don't know what device this is, though
@ebaauw The new Lutron Aurora Friends of Hue switch has two sensors on a Philips Hue bridge. A ZLLSwitch and a ZLLRelativeRotary sensor.

cobch, 2020-02-26:
Hey @KennethLavrsen 

Any news about the strange behaviour of how names are managed for the Magic Cube?

When I add the cube I got all these different alternatives without the possibility to choose. It seems to happen randomly )-:

1- two different names:
‘Mi Magic Cube’ And ‘ and ‘lumi.sensor_cube.aqgl01’ (Only for rotation event)

2- two different names:
‘Mi Magic Cube’ And ‘ and ‘switch-38’ (only for rotation event)

3- only one name:
‘Mi Magic Cube’ (for all events included rotation). These works really fine to manage all sort of event and should be the best way to handle the cube!

I have to change my automation each time the cube is disconnected (which happen quite often when I restart Hassio).

KennethLavrsen, 2020-02-26:
I reported the issue. I have not seen any notes from the developers that they have even touched this one.

Mimiix, 2020-06-05:
@KennethLavrsen is this still a issue?

closed 2020-06-12 by Mimiix

johnkread, 2020-08-02:
This is still an issue.

Mimiix, 2020-08-02:
@johnkread instead of replying that it still is a issue after it was closed, could you please open a new one with your issue? This makes it easier to solve :)