dresden-elektronik/deconz-rest-plugin #2038: EnOcean Senic friends oh hue switch

rendgeor, 2019-11-05:
The switch can be connected after some tries (Channel15, upper left button).
Unfortunately it is not shown in deconz gui.

The key mapping in phoscon works but the assigned action does not happen:
[image]

rendgeor, 2019-11-05:
@manup How can the  Request-device-support be done if the device isn shown in the deconz gui?
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

bedaes, 2019-12-07:
Same problem with my Senic switch. Did you already solve it?

bedaes, 2019-12-07:
Nevermind, I missed step 5 from here: https://phoscon.de/en/support#pairing-friends-of-hue-switch

rendgeor, 2019-12-17:
Update: Short key presses work if assigned directly within a lamp group.

But how can long press events defined?
[image]

closed 2020-04-23 by stale[bot]