dresden-elektronik/deconz-rest-plugin #2039: How Reset Hue strip for inclusion to new network

StoricU, 2019-11-05:
I have had a lot of trouble with my previous setup, so now I have started totaly from scratch.
Bought the new ConBee 2 and fresh install on the RPi.

I have managed to include all my IKEA bulbs and remotes, but just can't get the Hue Light strip included.
The IKEA has a easy reset method, but have been searching all over the web for how to reset Hue lights. So that I can include it to my new setup.

Is there anyone that have any information that could help me?
I'm really stuck..

ebaauw, 2019-11-05:
Hue lights cannot be reset using a power cycle.  The Hue bridge will reset them, when you remove them from the Hue app, but only if the device is powered and reachable.  Otherwise you need to reset them over touchlink.  Make sure to power-cycle them before touchlink resetting them.  The recent firmware versions only accept touchlink for some minutes after powering up.

The easiest way to do a touchlink reset is by using a Hue dimmer switch.  Press and hold the _On_ and _Off_ buttons for ~10 seconds, while holding the dimmer close (< 10 cm) to the light.  Hold until the light stops blinking.

If you don't have a Hue dimmer switch, you could try the old web app (under _Settings_).  If might be challenging to bring the RaspBee or ConBee close (< 10 cm) to the light.  Power the Raspberry Pi from a USB power bank for the RaspBee, or use a long (up to 5m) USB extension cable for the ConBee.

StoricU, 2019-11-06:
> Hue lights cannot be reset using a power cycle. The Hue bridge will reset them, when you remove them from the Hue app, but only if the device is powered and reachable. Otherwise you need to reset them over touchlink. Make sure to power-cycle them before touchlink resetting them. The recent firmware versions only accept touchlink for some minutes after powering up.
> 
> The easiest way to do a touchlink reset is by using a Hue dimmer switch. Press and hold the _On_ and _Off_ buttons for ~10 seconds, while holding the dimmer close (< 10 cm) to the light. Hold until the light stops blinking.
> 
> If you don't have a Hue dimmer switch, you could try the old web app (under _Settings_). If might be challenging to bring the RaspBee or ConBee close (< 50 cm) to the light. Power the Raspberry Pi from a USB power bank for the RaspBee, or use a long (up to 5m) USB extension cable for the ConBee.

Thanks for your reply!

I don't have a Hue dimmer switch, but will try to use the WebApp with an USB extension.

StoricU, 2019-11-06:
Just tried this, but the gateway could not find the light.
Even though I had the gateway <10cm from the light.
[image]

Also tried to unplug the light and then do a reset, but the gateway could not find it.
[image]

Something I'm missing?
Is there a channel limitation on the Hue?
Due to a lot of activity on the other channels I changed to 25.

ebaauw, 2019-11-06:
Afaik, touchlink should find the light regardless of the ZigBee channel.

Are you able to find other (types of) lights with this setup?

StoricU, 2019-11-06:
> Afaik, touchlink should find the light regardless of the ZigBee channel.
> 
> Are you able to find other (types of) lights with this setup?

Yes, I have 5 IKEA lights added.

StoricU, 2019-11-06:
Touchlink does not work on IKEA lights either.
Nothing comes up when scanning.

ebaauw, 2019-11-06:
Not that I know of, but it not a function that’s used very often.

StoricU, 2019-11-08:
Bought a HUE dimmer switch and got it reset. Included nicely to deconz now.

closed 2019-11-08 by StoricU