dresden-elektronik/deconz-rest-plugin #2040: [Question] Symfonisk events

lbouriez, 2019-11-05:
Hello, I am trying to use the Symfonisk remote using events in appdaemon.
I can't find a way to handle the volume correctly since I get just 1 event to start and 1 event to end.
Did someone managed to handle the change of volume with the remote ?
My actual appdaemon module is:
```
import appdaemon.plugins.hass.hassapi as hass

class SymfoniskSonos(hass.Hass):

    def initialize(self):
        self.sonos = self.args['sonos']
        if 'event' in self.args:
            self.listen_event(self.handle_event, self.args['event'])

    def handle_event(self, event_name, data, kwargs):
        if data['id'] == self.args['id']:
            if data['event'] == 1002:
                self.log('Button simple click - toggle')
                self.call_service("media_player/media_play_pause", entity_id = self.sonos)
            
            if data['event'] == 1004:
                self.log('Button double click - choose source')
                self.call_service("media_player/select_source", entity_id = self.sonos, source = self.args['source'])
            
            if data['event'] == 1005:
                self.log('Button triple click - next track')
                self.call_service("media_player/media_next_track", entity_id = self.sonos)

            elif data['event'] == 3001: # 3003 = stop
                self.log('Button volume up')
                self.call_service("media_player/volume_up", entity_id = self.sonos)

            elif data['event'] == 2001: # 2003 = stop
                self.log('Button volume down')
                self.call_service("media_player/volume_down", entity_id = self.sonos)

```

ebaauw, 2019-11-06:
This is still a mystery to me.  Afaik, Sonos doesn’t have volume change start and stop commands.  I need to pair the remote to my Trådfri hub and sniff the IP network traffic to see what Sonos commands the hub sends to the zoneplayer.

noxhirsch, 2019-11-06:
@lbouriez Unfortunately, I didn't have the time yet to do it, but that's the plan in my mind:
AFAIK with slow rotation the controller sends lots of start/stop signals. I'd check what's the longest delay between those is, before it only sends one start and a stop at the end of rotation. E.g. 500ms
Then I'd put in the script that when there's no stop signal within these 500ms it loops the volume up/down signal (plus or minus 2% volume) until the stop signal comes. So if you rotate for 2 seconds it sends 5 volume up/down signals to SONOS (one at the beginning and then four in the loop)

lbouriez, 2019-11-06:
@ebaauw Thx :)
@xhirschirsch Thx, seems to be an alternative, I will try tonight

stoneobscurity, 2019-11-06:
just have it +5 for every 3003 and -5 for every 2003. that's what i'm doing in Node-Red.

lbouriez, 2019-11-06:
But like this you can't just turn the wheel and expect the volume to slowly increase :(

stoneobscurity, 2019-11-07:
it sends multiple 3003 if you keep turning SLOWLY in my testing. a 1/2 CW turn got me like 4x 3003s.

ebaauw, 2019-11-08:
Managed to get the controller working thru the Trådfri gateway.  It seems to send multiple Sonos commands to increase/decrease the group volume by 3%.  That seems to confirm the multiple 2003 or 3003 events while "mississippi" slowly turning the dial.

lbouriez, 2019-11-10:
I did this module in appdaemon that seems to work for now: https://github.com/lbouriez/hassio-home-assistant_config/blob/master/appdaemon/apps/modules/symfonisk_sonos.py
If anyone has a better idea, let me know, I am looking for the best implementation possible :)

curla92, 2019-11-10:
is there a method to use it to control sonos? im using button event via homekit with homebridge-hue plugin. thank y

ebaauw, 2019-11-10:
If you expose your zoneplayers to HomeKit using homebridge-zp, you can simply add the actions to the controller’s buttons (using Eve, as Apple’s Home app doesn’t support the custom characteristics homebridge-zp uses for change volume and change track).

lbouriez, 2019-11-11:
I will close the issue since I solve all my questions :)

closed 2019-11-11 by lbouriez