dresden-elektronik/deconz-rest-plugin #2044: [Device support request] konke switch

ultimat, 2019-11-06:
I've bought some KONKE switchs but it don't seem supported.

Here are the requested screenshots :
[image]

[image]

[image]

Tell me if you need more informations

ebaauw, 2019-11-06:
It seems to be supported alright (the node shows `Light 3`, suggesting it’s exposed as `/lights/3` by the REST API plugin).  It doesn’t provide power measurement; I don’t see what more support you would be looking for.

ultimat, 2019-11-07:
I'm sorry if I don't understand how it is supposed to work. The switch is recognized as a light, shouldn't it be recognized as a sensor to provide short/long press information?

In phoscon app it shows only as a light and not as a switch, that's why I've thought there's something missing.

ebaauw, 2019-11-07:
Generally speaking there's two types of switches:
- Wired in-wall switches (that replace 20th century in-wall switches), used to "smartify" dumb lights (and/or other devices).  Like smart plugs, these act as smart lights, allowing remote control of the wired light (or other device).  They expose server (blue) _On/Off_, _Level Control_, and/or _Scene_ clusters, that keep the device state.  The buttons control the wired output.  
These behave like smart lights, look (from a ZigBee level) like smart lights, and are exposed as `/lights` resources.
- Wireless switches (both in-wall as well as handheld), used to control smart devices (lights, smart plugs, or wired in-wall switches).  They expose client (grey) _On/Off_, _Level Control_, _Scenes_ clusters.  The buttons on these send ZigBee commands.  
These behave like sensors and are exposed as `/sensors` resources.

ultimat, 2019-11-07:
Thank you for your explanation.
The konke switch is wireless (only a small button), so it should expose _client_ clusters.
What am I missing?

ebaauw, 2019-11-07:
In that case, it should also expose a device type of _On/Off Controller_ (or something), not _On/Off Output_.  What device is this?  Do you have a picture, link, or description?  Does it have a battery?

ultimat, 2019-11-07:
You can find a description and some pictures here : [konke switch](https://fr.aliexpress.com/item/33015636907.html?spm=a2g0o.productlist.0.0.4c8bec0d0HS2qA&algo_pvid=5a04cdd0-a179-4ab3-b87d-043c5338003f&algo_expid=5a04cdd0-a179-4ab3-b87d-043c5338003f-0&btsid=53296f8c-26d9-4c98-95d1-f1f52d64c935&ws_ab_test=searchweb0_0,searchweb201602_4,searchweb201603_52)

ebaauw, 2019-11-07:
Great, yet another Chinese vendor that has found brilliant new ways to mutilate the ZigBee standard.

Does the light in Phoscon switch on and off, when you press the button?  The Xioami door sensor also uses a server _On/Off_ cluster to report wether the magnet is making contact or not.

ultimat, 2019-11-09:
I don't see any change when I press the button (short or long press)
[image]
The _OnOff_ attribute remains _false_.

ebaauw, 2019-11-09:
And if you create a binding from the _OnOff_ cluster to the coordinator and setup attribute reporting for _OnOff_?

ultimat, 2019-11-09:
I really thank you for your advice.
I see where the binding is, but I don't know what is the coordinator. Is it the conbee key ?
And how can I setup attribute reporting ?
Can you provide me a screenshot ?

venuel79, 2019-11-13:
i have the same problem, but i can turn on the sensor in Hass with the blue
[image]
 switch,and then when i push the button the "light"turns off. But when "light" is off, i cant turn on with button, only with the switch in Hass

closed 2020-03-19 by stale[bot]

HomeBro, 2020-04-09:
I have also a few Konke devices (a switch and a motion sensor) lying around that I can't use in Deconz :(

Any plans on supporting this brand in near future?

HomeBro, 2020-06-13:
> I've bought some KONKE switchs but it don't seem supported.
> 
> Here are the requested screenshots :
> [image]
> 
> [image]
> 
> [image]
> 
> Tell me if you need more informations

Hi, have you solved this problem? I'm still trying to get the Switch to work with Deconz/Homeassistant.

venuel79, 2020-06-13:
No i don't used becouse it's works only in one way...