dresden-elektronik/deconz-rest-plugin #2045: [Feature Request] Allow full remote access with headless deconz server

cputoaster, 2019-11-07:
I am very happy with the deconz functionality. Its the best generic zigbee support I have seen anywhere, so this is just nitpicking.
I am running the headless server because I don't want to install all of the huge, slow and potentially insecure X11 / vnc / desktop stuff on the server. Its also hard to make sure it always runs / starts correctly on reboot. In some debugging cases I do need more functions than the current phoscon app. So the current way is

- shut down deconz
- start deconz-gui from external X display
- wait for a long time until the zigbee network (100 devices) stabilizes (and hope the X11 connection does not break)
- not have full functionality in this time
- use the app for debugging (eg, the map, changing network id etc)
- when finished, stop the deconz-gui
- restart deconz headless
- wait again until stabilized

This is not very satisfying. Ideally you could offer some solution that can be used while deconz headless is running. Maybe
- Some way to attach the GUI app to a running deconz daemon
- Full support of all GUI features in the web app, including some real-time websocket-based svg map
- Support of all control features in the phoscon app, and allow snapshot export of map in some node format, eg UML that can be read by UML viewers

chimpy, 2019-11-19:
Huge +1

duffbeer2000, 2019-11-25:
If you use the docker container then you have a vnc connection to view deconz and you don't have the problem with the insecure x11 / vnc / desktop stuff because it's secure in the container and not on the host: https://github.com/marthoc/docker-deconz/

cputoaster, 2019-11-25:
Thanks, I saw this and might actually do it at some point. Are there any observations about the cpu / ram / performance overhead of the vnc-docker solution (running without vnc being actually connected, eg the standard scenario) vs headless?

andriej, 2019-11-25:
I have to restart docker after every vnc usage as it stays with CPU used for whole time with 10-15%

Smanar, 2019-11-25:
On my side I have a raspberry 3, with deconz + an home automation application with desktop, X11 and VNC, I m most of the time at 2/4 %. So for me the difference is from the devices number. Desktop or not don't make a big difference.

closed 2020-05-25 by stale[bot]