dresden-elektronik/deconz-rest-plugin #2046: Crash after update to .70

djwlindenaar, 2019-11-07:
I'm getting a Crash after upgrading to the .70 beta. I'm on a raspbian with the raspbee hardware. I get:

```
deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5SerialPort.so.5: no version information available (required by deCONZ)
deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Network.so.5: no version information available (required by deCONZ)
deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5: no version information available (required by deCONZ)
deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Gui.so.5: no version information available (required by deCONZ)
deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5: no version information available (required by deCONZ)
deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Gui.so.5: no version information available (required by /usr/lib/libdeCONZ.so.1)
deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5SerialPort.so.5: no version information available (required by /usr/lib/libdeCONZ.so.1)
deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5: no version information available (required by /usr/lib/libdeCONZ.so.1)
libEGL warning: DRI2: failed to authenticate
deCONZ: relocation error: deCONZ: symbol _ZN9QMetaType22registerNormalizedTypeERK10QByteArrayPFvPvEPFS3_S3_PKvEi6QFlagsINS_8TypeFlagEEPK11QMetaObject, version Qt_5 not defined in file libQt5Core.so.5 with link time reference
```

mysteryan, 2019-11-07:
GUI seems to crash for me as well on .70, but runs fine without GUI on Pi 3 / Stretch / Conbee.

manup, 2019-11-07:
Are you using the shipped plugin or is that compiled manually?

mysteryan, 2019-11-08:
Both for me.  Shipped .69 seems to be okay.  I noticed that it started crashing with a manually compiled one sometime after that.  Thinking it's an issue with my own compilation, I've just kept quiet about it.  With shipped .70, the crashing still seems to persist.

djwlindenaar, 2019-11-08:
As shipped. I noticed the name of the package changed to ....Buster.... from ...Jessie... 
Could that be the issue? I'm still on Jessie...

manup, 2019-11-08:
When installing via APT repository the Jessie package (also compiled on Jessie) should be selected. The steps in the installation instructions should setup the matching distribution:

https://phoscon.de/conbee2/install#raspbian

djwlindenaar, 2019-11-08:
I installed from here:
https://www.dresden-elektronik.de/deconz/raspbian/beta/

I guess that is now outdated... Maybe it's time to start using the apt route...

mysteryan, 2019-11-08:
> I installed from here:
> https://www.dresden-elektronik.de/deconz/raspbian/beta/
> 
> I guess that is now outdated... Maybe it's time to start using the apt route...

I also used that download link instead of using apt route.

salopette, 2019-11-08:
I also have the problem that Deconz has always restarted after a few seconds and was always asked for firmware update.

I have now reinstalled .67 and it works. Do not trust me to make an update again.

mysteryan, 2019-11-09:
Not sure about OP, but this issue seems resolved after using a manually compiled version of today’s commits.  GUI seems to runs as it should for me (via headless VNC).

salopette, 2019-11-09:
What is the best way to update Pi3 to the last beta?

Smanar, 2019-11-09:
Last deconz beta ? using package I think.
But if the V70 is not yet in stable branch after 15 days, I think there is a reason.

salopette, 2019-11-10:
How can you downgrade from .70 to .69?

manup, 2019-11-10:
Depends on the platform, assuming Linux you need to manually install the .deb package:

    dpkg -i <deconz deb file>

The files can be downloaded from:
https://github.com/dresden-elektronik/deconz-rest-plugin#precompiled-deconz-packages-for-manual-installation

salopette, 2019-11-10:
I have Pi3, is that right?

wget https://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.05.70-qt5.deb

sudo dpkg -i deconz-2.05.70-qt5.deb

manup, 2019-11-10:
Yes but to downgrade you need the .69 version.

salopette, 2019-11-10:
Ok, THX!

closed 2020-03-16 by stale[bot]