dresden-elektronik/deconz-rest-plugin #2050: [Feature Request] batteryLevel and lowbattery for Xiaomi lumi.sensor_ht or lumi.weather

frest-de, 2019-11-09:
Hi, is there any chance of getting the batteryLevel or lowbattery channel working for the Xiaomi / Aqara sensors?

ebaauw, 2019-11-09:
I don’t have the `lumi.sensor_ht`, but the REST API plugin does report the battery for my `lumi.weather` sensors.  What do you mean by “channels”?

Note that the button-cell-powered Xioami devices only report battery voltage.  deCONZ tries to translate that to percentage, but that’s inherently flawed.  The battery shows (nearly) full charge for a long time, and then drops rapidly.

Also note that the battery voltage is reported only thru the Xioami special attribute report, which is sent only once an hour.  The sensor tends to stop sending these reports on low battery, sometimes even before the low value has been reported.

frest-de, 2019-11-09:
Thanks for the fast and detailed answer. The REST API does provide these informations. I only looked it up in phoscon and the openhab binding, those does not show the batterylevel.  I read the long thread about the special attributes of these sensors in 2017 and thougth it would not work.
Why does the webapp don't show these informations? It works for the Xiaomi switches and motion sensors as well, both show the batterylevel.

ebaauw, 2019-11-09:
> Why does the webapp don't show these informations?

I don't know.  Phoscon is not open source.  For the openhab binding, best ask there.

frest-de, 2019-11-09:
I will do that, thanks for your support!

closed 2019-11-09 by frest-de