dresden-elektronik/deconz-rest-plugin #2051: [Device Support Request] SZ-SRN12N Sercomm Corp.

salopette, 2019-11-09:
Can you please add SmartThing siren!

[image]
[image]
[image]
[image]
[image]
[image]
[image]

salopette, 2019-11-14:
do you need more info?

Smanar, 2019-11-14:
I think it's enought, but no user motivated ^^.
You can try yourself.  You have 2 situations, the device is standard and you have already a siren that work like yours, so you just have somes lines to change on the code or you can have an exotic device and need to use so much workaround.

If you wana try your luck, I can explain you how to make modification on the code, but I can't promise you it will working.

salopette, 2019-11-14:
Hi, I think the device itself is problematic.
When I reset it and relearn it, it is connected for a short time and then the connection breaks off.

Smanar, 2019-11-14:
Ha ok, so not the same problem, harder.

closed 2020-03-20 by stale[bot]

salopette, 2020-03-20:
Why closed?
Please add the device.

Smanar, 2020-03-20:
It's a bot that close issue if no one is using it.

But ATM if the device can't be include in deconz, It's harder. You need something with a sniffer take a look on trafiic to see the problem

>When I reset it and relearn it, it is connected for a short time and then the connection breaks off.

beneffm, 2020-03-26:
Hi all!
I also have the device here, experiencing the same issue. Is there any chance to get it running? How can I help :) ?

paulmann1590, 2020-03-29:
The sirene is at the moment quite cheap available for 6 euros, so is there any chance to get it running? :)

SwoopX, 2020-04-16:
The device is supported out of the box as far as I can tell. However. for the pairing to be successful, it is required to apply a configuration change in deconz GUI. For that, go to _Edit -> Network settings -> Endpoints_ (or press F9 -> Endpoints) and add `0x0502` to the `Out clusters` of endpoint 0x01.

[image]

When the device has been paired, it is further (currently) required to manually enroll it to the coordinator. Again from deconz GUI, select the `IAS Zone` cluster of the device and enter the MAC address of your coordinator for attribute `IAS_CIE_Address` by double clicking on the value field (must be entered with a leading "0x"). If that is not done and the device looses power, it leaves the network and gets reset when connected to power again.

[image]

ebaauw, 2020-04-17:
> When the device has been paired, it is further (currently) required to manually enroll it to the coordinator.

@SwoopX you need to whitelist the device so a ZHAAlarm `/sensors` resource is created, for the REST API plugin will do the _IAS Zone_ enrollment.  Check the code for the Heiman siren.

SwoopX, 2020-04-17:
@ebaauw Pretty sure it did that during the tests, but let me double check it again. Thanks for the hint. There's always one more place to whitelist...

SwoopX, 2020-04-19:
> @SwoopX you need to whitelist the device so a ZHAAlarm /sensors resource is created, for the REST API plugin will do the IAS Zone enrollment. Check the code for the Heiman siren.

The reason actually was that the device is NOT IAS zone, but IAS warning device. The logic didn't handle that. It's fixed now and works smoothly.

Greglog, 2020-07-06:
Hey guys,

I am trying to get the siren working. Its visible in the Phoscon App, but I am not able to switch on the siren - neither directly in the app nor via OpenHab. 
The strange thing is: on the setting page, when trying to change the name, it goes on and make the alarm-sound
Any ideas? What could I check?

SwoopX, 2020-07-06:
It can be triggered via REST API only and works with values select and lselect on parameter alert for the lights resource.

Greglog, 2020-07-06:
Ahh - thanks
It works...