dresden-elektronik/deconz-rest-plugin #2052: Problem adding Develco Smoke detector (SMSZB-120)

tyr84, 2019-11-09:
Seems it's a common issue with theese that you have to restart deconz several times to get all the endpoints and clusters. I've previously been able to get everything added through several restarts and battery in/out on the develcos.

However with the current version (there has been several updates to deconz since i last added a SMSZB-120, so I'm unable to refrence a certain version which might have introduced a bug) I'm unable to add endpoint 26, which is the temperature sensor. It does not show up in the deconz gui, and "Read Simple Descriptor" does nothing.

I've tried deleting and reseting the device, and as mentioned, several restarts of deconz. ep 23 adds just fine, not 26

SwoopX, 2019-11-09:
Let me check that out. Sounds familiar, but managed to get it working.

SwoopX, 2019-11-10:
Ok, so here is what works for me:

- Reset the sensor and pull the battery out afterwards
- Search for new lights and insert battery
- After about half of the time, **in deconz GUI**, go to "permit join", set it to 0 and apply
- Immediately read the simple descriptors while the search for new lights hasn't ended.

Regardless of how silly this reads, it works and gives me all 3 endpoints in the GUI. Also, the light is created. However, due to whatever reason, I cannot get the sensors to be created any more...

Not sure what might have happened, as no changes specific to the device have been included since. Hope that helps a bit. If you figure out how to get the sensors back, lemme know.

ebaauw, 2019-11-10:
Damn, I saw the same issue earlier upgrading the firmware of my Trådfri remotes.  Somehow deCONZ v2.05.70 doesn't read the simple descriptors when the network is open.  See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1873#issuecomment-551502335.  Seems intermittent, though.  @manup, any idea?

SwoopX, 2019-11-10:
Make it at least .69 since this is my current test gateway version. But it didn't work out as well with earlier versions. Struggled with the Develco's since I started working with then a couple of months ago...

tyr84, 2019-11-11:
Nope, not working, still only getting cluster 23. Tried several times. 
I'm not too familiar with the technical details, but it seems that "read simple descriptors" doesn't work on this device. However when i pop the battery out and in, it announces it's endpoints.

There also seems to be some problem with deconz not immediately showing new endpoints. For cluster 23 it works to restart the device, then after restarting deconz it will show up.

SwoopX, 2019-11-11:
Out of curiousity, what date code do you have once the basic cluster has been read. I got one of the latest and probably one with the very first firmware, which behaves even worse...

tyr84, 2019-11-11:
date codes: 2018-11-21 08:19 and 2018-11-22 01:38

tyr84, 2019-11-11:
Although I got 4 others from the same week (22-26) that I've added previously with all clusters. This leads me to believe it might be a change in deConz causing the problem

SwoopX, 2019-11-11:
@tyr84 thanks. My older one has a date code from 2016 and does not seem to provide correct data with the node descriptor. With the newer one from end 2018, that's ok. Suppose that is a firmware issue there.

@ebaauw, @manup I also wasn't able to see any active endpoint request (and an automatic simple node descriptor request). Happy to support in testing if you have some ideas.

SwoopX, 2019-11-11:
#2043

Fr00t, 2019-11-14:
Looks like this might be what I'm struggling with as well. I've tried multiple times to add my smoke detectors, but it ends up in various devices/lights/sensors being added under various names (sometimes as Light XX, sometimes SMSZB-120, sometimes IKEA dimmer for some reason). 

I've got two sensors and I've tried removing both of them (resetting them, then pulling the battery). But i still see some "junk" (specifically cluster 23 IAS Zone device) left in Deconz. How can I get rid of that?

Running 2.6.69 of Deconz with latest Conbee II firmware.

SwoopX, 2019-11-17:
I might have found a fix for that behaviour. While trying to understand what is happening from the code perspective, I was originally after the binding issue. That lead me to pieces of code I haven't taken a look at before. Looking specifically at the possible debug output, I was wondering why that never occurred and it was a candidate "for yet another place to whitelist"

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/8bd724cef41aba17536acacb486355d0080e9ee2/rest_sensors.cpp#L2305-L2326

The function "handleIndicationSearchSensors" is entered, I got the first debug output and then nothing. After adding an entry for Develco and pairing a sensor, I got everything created as it should be. In deconz GUI, I almost immediately get all the endpoints automatically discovered and all the sensor nodes are created. Also, the debug output shows plenty of stuff which now makes sense. Fast probing is now entered, now quite sure what the difference is, but finally the active endpoints are queried or the corresponding responses are processed further respectively.

If I now can manage to find out why you still have to manually bind the power configuration and temperature cluster in deconz GUI to get proper attribute reporting, I'd be a happy man.

@ebaauw you might want consider giving it a go for the Trådfri remotes...

@Fr00t yeah, the names might change every now and then, but naming it as an IKEA device is completely new to me. Anyhow, I wouldn't care too much about them. Regarding the orphan devices, many people have that question lately. Right now, I'd say you'd have to delete them directly in the database but haven't done that myself yet. Might be a good idea to implement a feature any time soon which permantently deletes those devices after one month of disappearance or so...

tyr84, 2019-11-17:
Looks promising :-)

I'll have to wait for this to be implemented and rolled to hass.io to test it afaik

SwoopX, 2019-11-17:
Not necessarily. If you feel funny, you can manually compile the plugin from my fork (just included the change) and give it a go. I'd also recommend to back up the original plugin first before overwriting ;)

tyr84, 2019-11-19:
Seems I cannot. I'm running Hassio. Tried entering the docker container of deConz through ssh, and actually managed to compile your fork and install it. But it seems Hassio just wiped the changes afterwards. All my package installs are gone (qmake, make, g++ etc), and the libde_rest_plugin.so now has the same datestamp as the other plugins.

tyr84, 2019-11-19:
This is probably unrelated and due to me not understanding the mechanics of an IAS Zone, but when I trigger one of the alarms (either by test, or actually putting smoke through the detector) the zone status on that device will update accordingly. However the status does not spread to the other devices in the same zone. Are they supposed to behave like this?

SwoopX, 2019-11-21:
Already issued a pull request

Regarding the zone behaviour, I recall that should be possible, but I don't know what might be missing or required.

Btw, managed to have the automatic binding and attribute reporting working, but it is more like a dirty hack. Requires some further investigation of the root cause...

jukmanty, 2019-11-22:
@tyr84 Restarting add-on from hass.io creates container from original image wiping all the changes. I managed to change the plugin by copying compiled plugin into container and altering run.sh to copy compiled plugin over original lib from image. Container can be restarted without overwriting the changes using docker stop and docker start, but for some reason library does get loaded.

I also have Develco smoke detector and I'm planning to add it into my hass.io + raspbee + deconz setup.

tyr84, 2019-11-30:
@jukmanty Trying to implement your hack. The run.sh inside the container is not persistent. Where did you find the file to modify?

jukmanty, 2019-11-30:
```
ssh -i ~/.ssh/my_id_rsa -p 22222 root@hassio.local
login
docker exec -it addon_core_deconz /bin/bash
nano -w /run.sh
exit
docker stop addon_core_deconz
docker start addon_core_deconz
```

I can modify run.sh in the root folder to copy/create files etc. If container is recreated from the image then all changes to run.sh will vanish. This happens if add-on is restarted from Home Assistant UI and it also happens if there is error in run.sh. Hacking the container and testing changes this way can be quite tedious job. If anyone has better alternative testing procedures for raspbi + hass.io combination, please share 🙂

closed 2020-01-19 by tyr84