dresden-elektronik/deconz-rest-plugin #2053: Xiaomi Vibration Sensor registered as ZHASwitch instead of ZHAVibration

MadMonkey87, 2019-11-10:
According to the supported device list the Xiaomi vibration sensor should appear as ZHAVibration sensor with a vibration flag. Instead it appears as switch. Is that the correct behaviour?

ebaauw, 2019-11-10:
Old deCONZ version?

xhemp, 2019-11-10:
Interesting. I had 2.0.70 and saw a message on my advanced settings to update conbee's firmware. I did that and it reverted me to 2.0.67 which sees the vibration sensors as ZHASwitch, just updated to 2.0.69 , re-paired the sensors and they are now seen as ZHAVibration.

MadMonkey87, 2019-11-16:
The sensor was paired with an old version but still appeared as ZHASwitch on newer releases. re-pairing of the sensor worked, now it properly appears as vibration sensor

closed 2019-11-16 by MadMonkey87